Jyers/Marlin #563: Serial mismatch when PSU tries to connect to printer after powerup

jtannenba, 2021-04-22:
### Description

Fails with my setup...I use Octoprint and have a UPS on the pi also with APCUPSD running. Any one of the updates past Early March causes Serial mismatch when PSU powers the printer up and tries to connect. Would love to be able to use the new menu items.

### Steps to Reproduce

1. [First Step]  Tried cable
2. [Second Step]  Tried updating Octoprint
3. [and so on...] Tried going older on Firmware

**Expected behavior:** [What you expect to happen]
Expect connect to printer to be successful
**Actual behavior:** [What actually happens]
Serial mismatch.
#### Additional Information
By experimenting, v1.2.2 or later fails my setup.
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
Don't have these files, I've been using "Default 4.2.7. stock" from here.
Info from Octoprint:

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.128 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1903308800
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin bugfix-2.0.x (Mar 8 2021

cruzer619, 2021-04-22:
did you setup octoprint as stated in the wiki for this error ? https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

jtannenba, 2021-04-22:
yes on the first part.  
[image]  This sis a screenshot of my Gcode scripts.

cruzer619, 2021-04-22:
yah but did you configure the error handling part..?

jtannenba, 2021-04-22:
Note first line.  Yes

Jyers, 2021-04-22:
This seems to be an issue with the main Marlin branch at the moment, it may have been fixed but I need to do some further testing. Does it let you connect if you try again, or can you not connect at all.

jtannenba, 2021-04-22:
It would let me connect if I left the printer on and rebooted Octoprint.  Only by going back on the firmware am I able to connect with the PSU control in Octoprint.  Right now I'm running the Jyers 1.2.1 from 3/8/2021.  and yes, I believe the Marlin does this too.  I believe I tried several versions.  The one from Feb worked the new ones didn't.  I went back to this one because of the additional info on my display.  The v 1.2.2 says it had a fix for serial problems, and I guess that is what broke my setup.

twuytens, 2021-04-22:
I also experience the same issue with a fresh install of 1.3.1. Applied the 'error handling' steps but it really cannot connect.
```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600
Trying port /dev/ttyUSB0, baudrate 115200
Connecting to port /dev/ttyUSB0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x6d741f30, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyUSB0, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyUSB0, baudrate 230400
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyUSB0, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyUSB0, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyUSB0, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyUSB0, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Connection closed, closing down monitor
```

Let me know if I can provide some more info to help out.

Jyers, 2021-04-22:
@twuytens Your issue here is quite different. Octoprint is reporting no connection candidates whatsoever. I would encourage you to check your connections and such, as I have yet to see that kind of error caused by the firmware.

twuytens, 2021-04-22:
Well I did get the `serial mismatch` first, but apparently my printer was halted with M112 (due to that error?) and required a reboot too. After rebooting both the pi with octoprint and the printer it's working now. Apologies about the fuss, but thanks for the quick response. 😉

Jyers, 2021-04-22:
No problem, I'm just trying to figure this stuff out. Btw you can disable M112 shutdown commands from octoprint. Check the wiki page for the settings

jtannenba, 2021-04-23:
I am embarrassed. Somehow that send M112 go re-enabled with my testing. I shut if off again and saved it, went back to check and it was on again.  Shut it off and verified several times that it stayed off and tried the newest firmware.  So far, it seems to connect.

aryajuanda, 2021-04-23:
For me just i disable MEATPACK in firmware to resolve the issue, because i dont use meatpack plugin.. no more serial missmatch, becasue my raspi is up for 24h but not my printer, so idont have to restart my octoprint.. some people said this is not the solution, but its the best scenario for me.. 

Since in the original marlin MEATPACK is not enable by default, but is enable by default in this firmware, until the main route cause been fixed (some people said install meatpack plugin in octoprint seam fixed it but i dont like to use it) i will just disable MEATPACK support in the firmware.. changging octoprint behavior on this error it just like runaway from problem without fix it.. 

Again this is just my personal opinion.. i believe @Jyers has his own reason why he enable MEATPACK support by default when the upstream marlin firmware is not enable by default..

Jyers, 2021-04-25:
Are you still facing this issue? I've run my octoprint without meatpack (still enabled in firmware) for a week now to try to recreate this issue and have not seen a single issue.

jtannenba, 2021-04-25:
As above, I finally got the send m112 to stay off and loaded v 1.3.1 and so far, running good.

Jyers, 2021-04-25:
Glad to hear it. Feel free to update this if you face any more issues.

closed 2021-04-25 by Jyers

sengemann, 2021-05-05:
I just updated to `1.3.2` and am experience a similar issue to @twuytens . First I experienced a `serial mismatch` error. I rebooted my Octoprint and now it won't connect at all.

``` bash
2021-05-04 01:18:18,156 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 01:18:18,164 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2021-05-04 01:18:18,164 - octoprint.server - INFO - OctoPrint 1.5.3
2021-05-04 01:18:18,169 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Creality-2x-temperature-reporting-fix (0.0.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ender3v2tempfix
|  Custom CSS (0.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_customcss
|  Dashboard (1.18.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-05-04 01:18:18,187 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|      cores: 4
|      freq: 1200.0
|      ram: 915722240
|  os:
|      bits: 32
|      id: linux
|      platform: linux
|  plugins:
|      pi_support:
|          model: Raspberry Pi 3 Model B Rev 1.2
|          octopi_version: 0.18.0
|          throttle_state: '0x0'
|  python:
|      pip: 20.3.3
|      version: 3.7.3
|      virtualenv: /home/pi/oprint
2021-05-04 01:18:18,188 - octoprint.server - INFO - ------------------------------------------------------------------------------
2021-05-04 01:33:18,157 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 01:48:18,158 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 02:03:18,160 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 02:18:18,161 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 02:33:18,163 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 02:48:18,164 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 03:03:18,166 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 03:18:18,167 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 03:33:18,169 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 03:48:18,171 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 04:03:18,172 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 04:18:18,174 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 04:33:18,175 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 04:48:18,176 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 05:03:18,178 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 05:18:18,179 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 05:33:18,181 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 05:48:18,182 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 06:03:18,183 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 06:18:18,185 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 06:33:18,186 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-05-04 06:48:18,188 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... 207 lines cut ...]
2021-05-04 22:33:41,252 - octoprint.access.users - INFO - Logged in user: engemasa
[... cut ...]
| Recv: ÑaWÍ [kµÑ¡)Error:Serial status mismatch
| Recv: Error:Serial status mismatch
2021-05-04 22:33:52,267 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error"
2021-05-04 22:33:52,268 - octoprint.util.comm - INFO - Force-sending M112 to the printer
2021-05-04 22:33:52,290 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline after error"
2021-05-04 22:33:52,297 - octoprint.plugins.action_command_notification - INFO - Notifications cleared

```

Jyers, 2021-05-05:
First off check the wiki page on octoprint settings to disable M112. Have you tried resetting and storing eeprom since the upgrade?

sengemann, 2021-05-05:
I have not tried either of those yet. I will probably try this tomorrow and report back the result.

sengemann, 2021-05-05:
I have been able to resolve this issue. It would have been a good idea for me to first goto the printer and look at the darn screen so I could see the error message indicating me to power cycle the printer. That was probably the main issue as to why Ic couldn't get OctoPrint to connect. (Forgive me, I was gaming with some buddies at the time too 🤓).  I also disabled `M112` commands (blocked and removed it from the Emergency Commands input field). After doing this, I was able to connect to the printer successfully from Octoprint.

NOTE: I did NOT reset or do anything with the `eeprom`. Mainly because I've never done that before nor do I know how to do it.

Thank you for feedback and input! The Mesh Bed Leveling is AWESOME!!! 👍 👍 👍