Jyers/Marlin #569: BLTouch not applying mesh to prints?

Baillie521, 2021-04-23:
### Description

BLTouch doesn't use the saved mesh and doesn't apply it to the print.

### Steps to Reproduce

1. Turn printer on
2. Heat up to temp
3. Start print

**Expected behavior:** The saved mesh should be applied to the print.

**Actual behavior:** The mesh isn't applied, even when using M420 S1 and commands like M500 and M501?

#### Additional Information

I will list my start GCode along with my configuration files below. To be clear the BLTouch works, and the repeatabillity test shows that it is very accurate, it's just not applying any of the mesh to a print.

[Configuration.zip](https://github.com/Jyers/Marlin/files/6363825/Configuration.zip)

Start GCode:
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 ; home all
G29 ; BLTouch
M420 S1 ; Activate saved mesh
G1 Z2 F240
G1 X2 Y34 F3000
G1 Z0.28 F240
G92 E0
G1 Y214 E15 F1500 ; intro line
G1 X2.3 F5000
G92 E0
G1 Y34 E15 F1200 ; intro line

Bed Mesh Readout:
      0      1      2      3      4
 0 +0.026 +0.010 -0.011 +0.036 +0.098
 1 +0.024 -0.011 -0.031 -0.006 +0.042
 2 +0.014 -0.023 -0.059 -0.037 +0.002
 3 +0.020 -0.014 -0.027 -0.003 +0.042
 4 +0.032 +0.011 -0.008 +0.047 +0.087

Baillie521, 2021-04-23:
UPDATE: I have compiled a new firmware with the exact same config files as posted above but with the 1.3.1v to see if this helps. So far i have narrowed it down to the z-rod not compensating while printing?

Jyers, 2021-04-23:
Are you sure the mesh isn't being applied? Based on your mesh, your z axis will only be moving in the realm of a few hundredths of a millimeter. You might only really be able to notice it with a sweeping move across the whole bed. You also should make sure your are watching the z coupler very closely as that's the only way you will catch the movement.

Baillie521, 2021-04-23:
Thanks for the reply dude! I am certain that it wasn't moving, but now I have done that firmware change and changed my start GCode so it reads as below, I can now feel the lead screw moving slightly but when I do a be level test the left side of my bed is slightly too close and the right two far away leaving gaps in the little circles? I just can't seem to get the bed levelled because of this and it's annoying lmao, I would've thought the BLTouch would've compensated for a bed with the readings as above?

Start GCode:
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 ; home all
M420 S1 ; Activate saved mesh
G29 ; BLTouch
G1 Z2 F240
G1 X2 Y34 F3000
G1 Z0.28 F240
G92 E0
G1 Y214 E15 F1500 ; intro line
G1 X2.3 F5000
G92 E0
G1 Y34 E15 F1200 ; intro line

Jyers, 2021-04-23:
If it's compensating then everything should be working properly. The only thing I can think would be is your z leadscrew has too much drag or a bit of backlash preventing it from compensating correctly. I'm closing this as it's not a firmware issue, but feel free to keep updating this with any findings and I'll try to help figure out the issue.

closed 2021-04-23 by Jyers

Baillie521, 2021-04-23:
Thanks for your help, it seems that you may be onto something there as I can see that the leadscrew is moving but its like its not compensating much if not at all. My reasoning for this hypothesis is because when I do a bed level test the squares on the left of my bed appear too squished yet on the right they hardly adhere to the bed, even after thorough cleaning and even flipping the glass plate?

Baillie521, 2021-04-23:
I just need pointing in the right direction lmao

Jyers, 2021-04-23:
Check your z axis leadscrew. It should be very easy to turn the coupler by hand.

Baillie521, 2021-04-23:
@Jyers yep, this is very easy to turn :)

Jyers, 2021-04-23:
Hmmm, you could try a more detailed mesh, either by using one of the UBL versions, or compiling your own with a larger mesh size.

Baillie521, 2021-04-23:
So I'm using the BLTouch and it seems that it won't execute the mesh as it should and I don't know why, this was ever since I got my Hemera for the printer?

Jyers, 2021-04-23:
If it started after installing the Hemera it could be the added weight limiting your steppers ability to adjust the height as accurately, but thats just my best guess.

Baillie521, 2021-04-23:
Yeah, that's not a bad shout, I wouldn't even know where to begin to sort that out, I don't want to sacrifice a nice level bed because if I do having a nice expensive hemera isn't really worth it lmao