Jyers/Marlin #571: V1.3.1_E3V2_ManualMesh 5x5 v4.2.2 - printer not remember current Z when resume after pause and turn off power overnight

shoppingcheck, 2021-04-23:
Description
I use the printer for prototype at work so I must turn off before leaving the office due to safety concern. After installing the latest firmware v1.3.1, I found out that the printer does not remember current Z when resuming the print after pause and turn off power overnight. When resume printing, the printer print with Z above the current print (see pic)

Steps to Reproduce

1. After this issue happened, I try to verify by print another one, I pause in the middle of printing, turned off power. After 5 minutes, I turn on the power and resume print, it prints correctly with current Z. 
2. So I re-starts to print my product again, pause print, turn everything off before I leave office at 5pm. Next day morning, I came in and resume the print, then it happened again, not continue on the current height.
3. I'm not sure what cause the problem since I'm new to 3D printing. Due to safety restrictions, I must turn off the printer before go home. This issue cause me hard time since most of my works are take really long time to print, mostly over 12 hours.

Expected behavior:
Continue to print with current height when resume.

Actual behavior:
Printer print above current height when resume printing.

Additional Information
This is the picture I took showing it print above current height. I trying to print a cable chain as a sample to show the issue.
I let it print for 5 mins after resume and already clean lot of messy filament before taking the picture.
[image]

Jyers, 2021-04-23:
Interesting, I'll give this a try to see if I can recreate it. Most likely you'll want to post this issue on the main marlin repo though, as it would be an issue with the power loss resume feature, which is not in my jurisdiction to fix.

shoppingcheck, 2021-04-23:
Hi Jyers,

Thank you for your reply. 
I didn't have this issue when using manufacture firmware which comes with the printer.
Like many others, I have a warped bed and been trying to levelling the bed but not successful so far, maybe I didn't do it right.
Your firmware helps me overcome that issue by allowing me to adjust the Z offset in different areas of the bed. 
I've got rid of the bed leveling issue after using your firmware before I ran into this bug. 
So I thought it would be a great idea to reach out to forum and see if anyone gets the same issue but found none in closed issues.

Thank you and have a great weekend!

tititopher68, 2021-04-25:
This issue still unofficially exists on Marlin, the problem would be related to MicroSD readers and the faulty recording of the PLR file (majority opinion on the subject) ...
Some even reused Marlin 2.0.6.x versions of powerloss.cpp and powerloss.h to get proper operation. But in others, it does not change anything ...
On other sites, we say that you have to uncomment these 2 options in Configuration_adv.h and put the indicated values:
#define POWER_LOSS_PIN         -1
#define POWER_LOSS_STATE      LOW

Personally, I turn off this feature which quickly destroys MicroSD cards.

Jyers, 2021-04-25:
Closing this for now as there is not much I can do. If you continue to face the issue and have any new information feel free to update this issue and I would be happy to try to help.

closed 2021-04-25 by Jyers