Jyers/Marlin #576: Cursor selection does not respect selected position

AngleV, 2021-04-24:
relates to this https://github.com/Jyers/Marlin/issues/548

When you select a menu option that brings up a Popup and you choose Cancel then the selection does not stay in the same position that you were before.

Steps to reproduce.
1. Choose Prepare --> Change Filament.
2.  Popup appears.
3. Choose Cancel
4. Menu redraws but selection now goes all the way up to Load Filament.

IMHO it should stay in the Change Filament.

tititopher68, 2021-04-24:
That's right. In the code, for Popup, the selection variable is set to 0 to be able to select: Confirm or Cancel.
so we should replace in the part of the code corresponding to Prepare - Change Filament, Redraw_menu () by Draw_Menu (last_menu, last_selection);

Jyers, 2021-04-24:
Hmmm, yes @tititopher68 is right. I created last_selection for this purpose, but must have forgot to implement it.

tititopher68, 2021-04-25:
I just tested and it's OK for all issues from AngelV. 
I change also Redraw_Menu() in case ConfFilChange with Draw_Menu(active_menu, last-selection)

Jyers, 2021-04-25:
This should be fixed now I believe

closed 2021-04-25 by Jyers