Jyers/Marlin #579: Change Filament during print goes nuts

AngleV, 2021-04-24:
1. Start a print.
2. Select Tune --> Change Filament
3. Select Cancel

Chaos appears :-)

[image]

Jyers, 2021-04-24:
Oh my.... Must be an issue with the popup I added. I'll get this fixed.

tititopher68, 2021-04-25:
fixed with Draw_Menu(active_menu, last_selection) in case ConfFilChange

Cal-LF, 2021-04-25:
Also found today that the filament change in my compiled manual mesh 3X3 does not work properly any more. There is a 40 -50mm filament unload before the print resumes.

Jyers, 2021-04-25:
@Cal-LF That's how it has always been, you can tune those parameters in the advanced pause section of configuration_adv.h

tititopher68, 2021-04-25:
@Cal-LF this is not the subject here ;-)
In your case, you have to go to the Advanced menu (Main Menu -> Control Menu -> Advanced) and modify the Load Lengh and Unload Lengh parameters. Add to the displayed values, for example, 60 to 70 mm. And then don't forget to do a Store Settings (Control Menu).

Jyers, 2021-04-25:
I believe fixing #576 should have also fixed this

closed 2021-04-25 by Jyers