Jyers/Marlin #582: Uncomment #define PRINTCOUNTER by default?

Legsmaniac, 2021-04-24:
### Description

I keep getting asked on many forums I participate in how others can get the same information on the Info screen that I have.  I post similar photos to this photo on some forums sometimes and it gets heads scratching.

[image]

I reply (over and over) that you need to edit the `Configuration.h` file and uncomment `#define PRINTCOUNTER`, recompile and flash printer.

My question is - why is this line not uncommented by default as I think it's a great and useful feature and doesn't do any harm by leaving it uncommented by default.  Unless I'm mistaken?
If indeed there's no harm, can this line be uncommented in all future versions?

gothcha, 2021-04-24:
Using 1.3.1 if I uncomment PRINTCOUNTER the info display gives 0 for everything. Does it only start to total the data after it has been enabled?

Legsmaniac, 2021-04-24:
To be honest, not sure.
Having said then, when I discovered it and enabled it a few months into my new printer, it already showed several past print jobs so I'm guessing the information must be stored in the mainboard computer somewhere.  So I would have thought yours would show several past print jobs too.  Maybe it's different with different printers?  Maybe this is why this wasn't enabled by default?  Perhaps Jyers will know more?  My printer is an Ender 3V2 with Creality 4.2.7 mainboard.

gothcha, 2021-04-24:
Mine is the V2 with 4.2.2. I will see if it starts to accumulate data from now on. Have you used the printer with Octoprint? It suggests that this responds to M73 command to report print info.

Legsmaniac, 2021-04-24:
Nope.  Never used Octo.  I use ESP3D instead.  Much prefer it.
Updated to latest just now to check all was still well.  😊

[image]

It's even carried forward the last extra two prints I did since the last photo.

Jyers, 2021-04-24:
You're right, I'll enable this for future releases

closed 2021-04-24 by Jyers

duncio, 2021-04-25:
> Mine is the V2 with 4.2.2. I will see if it starts to accumulate data from now on. Have you used the printer with Octoprint? It suggests that this responds to M73 command to report print info.

I also have 4.2.2 and it works as expected, showing all the stats.

gothcha, 2021-04-25:
Thanks, it is displaying the stats OK from the point that it was first enabled, so maybe for you it had been enabled in an earlier firmware so may have have continued to collect data.
Still will be useful going forward

Legsmaniac, 2021-04-25:
I can confirm mine was never "enabled" yet was storing all the stats, if by enabled meaning print counter was uncommented.  All printers store stats, even inkjet printers.

Ali229, 2021-04-25:
> You're right, I'll enable this for future releases

Not sure if intended or not but this isn't enabled for default config.

Jyers, 2021-04-25:
> Not sure if intended or not but this isn't enabled for default config.

@Ali229 I don't modify the default config to keep it fairly stock and to avoid messing with peoples configs when the pull to their personal repos. All configs are built off of the templates in the configuration files folder