Jyers/Marlin #604: Print screen

jfrenelle, 2021-04-26:
The print screen goes away and show the home screen

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

smooreace, 2021-04-26:
Just have to LOL

Jyers, 2021-04-26:
Any further details would help here. When is this happening, what kind of printing (SD/Host), when did this start, is it repeatable. There is nothing I can really do with "it did this, help"

jfrenelle, 2021-04-26:
It started on the first print
It's on a new ad card
It will finish the print 
Ender3 v2 
Jjyers 1.3.1 
It was April 25 2021

Jyers, 2021-04-26:
When in the print does the print screen go away?

jfrenelle, 2021-04-26:
Right when it go to print the purge line after it zeros out

Jyers, 2021-04-26:
Can you post your start gcode

jfrenelle, 2021-04-26:
G28
M501
M420 s1 z2
[image]

Jyers, 2021-04-26:
Does this occur for all gcode files you have tried?

jfrenelle, 2021-04-26:
It's the same gcode as I run on the Jyers 1.3.0
I use the same slicer and setting on both of then
No problems on 1.3.0 runs great

jfrenelle, 2021-04-26:
Both are ended 3 v2

Jyers, 2021-04-26:
Try resetting eeprom by going to the control menu and then selection reset defaults and then store settings

jfrenelle, 2021-04-26:
Its printing now so when it's done I'll give it a try and let you know 
Thank you

jfrenelle, 2021-04-27:
Ok i reset the reproduce and the same thing

jfrenelle, 2021-04-27:
[image]
[image]
[image]

closed 2021-04-27 by jfrenelle

jfrenelle, 2021-04-27:
I rest and no bed leveling at all 
Just printed
Now I will bed level and try again

jfrenelle, 2021-04-27:
This is the HS 1.3.1

jfrenelle, 2021-04-27:
4.2.7 main board

jfrenelle, 2021-04-27:
Same thing after auto leveling and printing

Jyers, 2021-04-27:
Very strange. Do you have any kind of host device connected to your printer? Octoprint, pc, etc

jfrenelle, 2021-04-27:
Nope just a sd card

Jyers, 2021-04-27:
That is so strange. Other than the gcode or a hardware issue I can't imagine what would cause that.

jfrenelle, 2021-04-27:
I may just flash it back to 1.3.0

jfrenelle, 2021-04-27:
Thank you for trying

Jyers, 2021-04-27:
Let me know if it does go away in 1.3.0 because that would at least be a start

jfrenelle, 2021-04-27:
Yes it goes away with 1.3.0

jfrenelle, 2021-04-27:
I mean the problem goes away

jfrenelle, 2021-04-28:
If you need a test subject send the firmware to me I'll flash it and try it for you

Jyers, 2021-04-28:
The main issue here is you're the only person I've heard from experiencing this problem, so I don't even know where to start here... I'll let you know as soon as I can think of a possible cause for this.

Tuncay-Ayhan, 2021-04-28:
> If you need a test subject send the firmware to me I'll flash it and try it for you

I'm not sure, but you could try to change your start code to what I have.

`; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
M190 S{material_bed_temperature_layer_0} ; preheat the bed
G28 ; autohome
G29 L0 ; load mesh from first slot
G29 J5 ; make autotilt for 5*5 grid
G29 F10 ; set Z fade to 10mm
G29 A ; enable UBL
M104 S{material_print_temperature_layer_0} ; set extruder temp
M105
M109 S{material_print_temperature_layer_0} ; wait for extruder temp

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish`

jfrenelle, 2021-04-28:
Found the problem
Took out the M501 out and the screen stayed just like normal

Jyers, 2021-04-28:
Ahhhh that makes sense. M501 redraws the screen, but I don't think its coded to work on the print screen. I'll probably set that up to fix it.

Alfagek, 2021-05-09:
> Ahhhh that makes sense. M501 redraws the screen, but I don't think its coded to work on the print screen. I'll probably set that up to fix it.

@Jyers Jake did you fixt this problem in 1.3.2? Because i also have this problem with 1.3.2. with the 4.2.7 board. It's not always stays on the mainscreen. 

What i did know, it's happen to me, when i start more prints after one succesfull print. Working with octoprint. First print i do get the print menu, after that it stays on the main screen.

Did all ready try to reset setting en store settings, but didn't solve the problem. Also did not try to flash back a older version, to see if the solve the problem(busy printing motherday gift).

Keep up the good work (y)

Jyers, 2021-05-09:
I don't think I ever fixed this, I'll put it on the list this time to be sure it gets done soon.