Jyers/Marlin #606: Setting a message with M117 then opening Tune clears the message upon return to the main Print menu

nullstalgia, 2021-04-26:


Jyers, 2021-04-26:
I feel like there was a reason I did this, but I'll check the code to see if I can keep the message through menu changes.

Jyers, 2021-04-26:
I think I've fixed this now

closed 2021-04-26 by Jyers

nullstalgia, 2021-04-27:
@Jyers Can't compile, 

AUTO_BED_LEVELING_UBL is defined. 

[image]
[image]

Jyers, 2021-04-27:
Oops, that needs to be ENABLED(AUTO_BED_LEVELING_UBL) && !HAS_BED_PROBE. I'll fix that now

Jyers, 2021-04-27:
Alright, try now.

nullstalgia, 2021-04-27:
Compiled! I'll test it out once this print finishes (couple hours or so). Thanks again.