Jyers/Marlin #614: Extruder turns in retract direction on extrude and vice versa

laebshade, 2021-04-26:
### Description

On an Ender 3 v2 with 4.2.7 board and Jyer's v1.3.1 E3V2-BLTouch-5x5-HS-v4.2.7.bin firmware, the extruder turns in retract direction on extrude and vice versa.

### Steps to Reproduce

1. Flash https://github.com/Jyers/Marlin/releases/download/v1.3.1/E3V2-BLTouch-5x5-HS-v4.2.7.bin
2. Heat nozzle to 215c
3. Try to extrude via LCD or via Octoprint terminal commands

**Expected behavior:** extruder motor rotates clockwise to push out filament

**Actual behavior:** extruder motor rotates counter-clockwise, retracting filament

#### Additional Information

I was able to flash a compatible firmware from https://marlin.crc.id.au/ and extrude correctly.

I previously used Jyer's on the 4.2.2 board with no such problems.

Jyers, 2021-04-26:
Is your extruder setup stock? Also does this issue also pertain to prints or just the move menu?

laebshade, 2021-04-26:
> Is your extruder setup stock? Also does this issue also pertain to prints or just the move menu?

Yes. I am using the stock cable and stock extruder.

It happens anytime I try to extrude: whether from a print, the move menu, or the "extrude" button in Octoprint's control tab.

Jyers, 2021-04-26:
This most likely means your extruder is reversed from the stock direction. You can either fix this by compiling your own version of the firmware or switching the wires on your extruder connection

closed 2021-04-27 by Jyers

laebshade, 2021-04-27:
It's not reversed. I tried switching around wires, and that just caused the
extruder to stop working. I tried compiling your firmware and changing the
bool flag to run in reverse; it did not work (extruder wouldn't move at
all).

I mentioned in the original bug report I was able to use a different
firmware without issue.

I'll pass on custom compiling and stick with a different firmware that
works for the 4.2.7. That's a shame, because I really like the features you
offer, but the extruder not working correctly on your compiled is a deal
breaker.

Jyers, 2021-04-27:
Just to check you modified this? `#define INVERT_E0_DIR false`

The way this firmware handles motors and directions is the same as base marlin so there should be no difference between this and any other compiled version.

laebshade, 2021-04-27:
Yes, that is the same value I flipped to true.

I agree there should be no difference. I have no explanation on why your
firmware doesn't work while the marlin crc au version does.

Jyers, 2021-04-27:
Have you trued reversing the other extruder direction, in the case that maybe your extruder motor isn't plugged into the E0 pin?

laebshade, 2021-04-27:
> Have you tried reversing the other extruder direction, in the case that

You mean retracting? No, I don't recall that I did.

Jyers, 2021-04-28:
No, I mean like setting `#define INVERT_E1_DIR` or the others to true rather than just E0

laebshade, 2021-04-28:
No I have not.

However since I was able to use another firmware, and you've closed the
issue, I'll set that aside as an option for the future.

Jyers, 2021-04-28:
Okay, whatever works best for you. I closed the issue as it seems to be an individual config issue rather than a general firmware issue. That being said, feel free to update this with any new findings as I would absolutely like to help you solve this issue.