Jyers/Marlin #625: Is this Marlin fork still board agnostic?

Fredyy90, 2021-04-27:
Hi,

Thank you for your great work for the community.
But as you changed a lot in the firmware i'm unsure if'll still be able to use it with a different mainboard than the creality 4.2.2 or 4.2.7.
I'm currently searching for a board with at least 5 uart connected tmc2209 stepper drivers to create a dual-z setup with 2 steppers that sync via my bltouch and implement linear advance. is something like this still possible or do i have to go back to mainline marlin and loose all the new features you integrated as you changed to much specific to the stock hardware?

smooreace, 2021-04-27:
The steppers can be run in parallel from one controller quite well with a
4.2.2 board. I've been thinking about picking up a 4.2.7 as a backup, but
seems like a waste of $$. And, it still has the same setup requirement

Fredyy90, 2021-04-27:
> The steppers can be run in parallel from one controller quite well with a 4.2.2 board. I've been thinking about picking up a 4.2.7 as a backup, but seems like a waste of $$. And, it still has the same setup requirement
> […](#)

sure they can be run in parallel, but what stops them from getting out of sync over time? when the machine is powered off or moved manually? 
That's why I want to go the 2 driver route to tram the X-gantry to the bed via my bltouch before every print and thats why I'll need to change the board.
Marlin does this already, so the feature should also be available here, but I don't know if this fork of marlin is still compatible with other boards while keeping all the ender3 v2 specific changes and display code.

Jyers, 2021-04-27:
This fork actually has zero modifications to the marlin code outside of the display portion. So it will be fully compatible with anything you might want to run it on 👍

closed 2021-04-27 by Jyers