Jyers/Marlin #628: zero all the axes

Lighty1989, 2021-04-27:
Hi Jyers, you made great firmware. Unfortunately I have to say that the original option is no longer available where you can zero all the axes. this would be beneficial if some use a laser. would be great if you could add it right away. Greetings from Germany

Jyers, 2021-04-27:
Where was this option in the original firmware? If you want to change the zero position of the printer, you can use the home offsets settings in control>Motion

Lighty1989, 2021-04-27:
There was a point in the original firmware where all axes are zeroed to one position.

that was about home axes

tititopher68, 2021-04-27:
I think he says : Home X, Home Y, Home Z. With a Laser Head, they only use X and Y axis...

Lighty1989, 2021-04-27:
for the laser you also need the z-axis for zeroing

you can add an option especially for the 3 axes with one click

at least 50mm + the thickness of the serious object

Lighty1989, 2021-04-27:
[image]
 This is how it looks in the original firmware, and if you set a position in x y and z at move you can click at Set home offset and all axes will be set to zero at this point

Jyers, 2021-04-27:
@Lighty1989 Ahhh, I see. You can still do this with the firmware, but it's not quite the same approach. In `Control>Motion>Home Offsets` there are options to set the x and y home offsets. However this does not offer the ability to set the z offset like this. I will look into adding an item to do set the current position to 0 on all axis like you described. It will most likely go into the home offsets menu.

closed 2021-04-27 by Jyers

Lighty1989, 2021-04-28:
If you could do this, maybe relatively soon, I would be very happy.

thank you, it is a very important function for me, as i also use a laser as I said

kind regards
Sven

Lighty1989, 2021-04-29:
[image]

Is that this function by chance?

how long could it take you to add this menu?

i think your work is great :-D

tititopher68, 2021-04-29:
@Lighty1989 : yes, it is indeed this function M428 or M206.  On the other hand, this requires reviewing its ergonomics.  Indeed, M428 alone is radical.  Like that, in the factory firmware, to go back, you have to do a complete Reset Settings of the printer.  Here it would be necessary to be able to return to the initial parameters without having to do a Reset to default ...

tititopher68, 2021-04-29:
@Lighty1989 : I implemented this functionality in my last build V1.4.3.a.
You can set the offsets directly in the dedicated menu and reset to the normal position without having to reset the printer to default.
Firmware to compile here : https://github.com/tititopher68/Marlin

Lighty1989, 2021-04-30:
Nice, is your firmware based of Jyers, lastest Update?

tititopher68, 2021-04-30:
@Lighty1989 : Yes with a lot of additions and some fix.

Tuncay-Ayhan, 2021-04-30:
> @Lighty1989 : Yes with a lot of additions and some fix.

Non-issue related question.

I'm quite new to github stuff, so I was wondering how do you merge the latest updates from Jyers/Marlin into your fork/branch?

Im trying to maintain a fork that is parallel to Jacobs, with my custom settings and also up to date with his latest additions.

I've read about it, watched videos about, but still don't understand it.

**Is this the right way?**
[image]

Lighty1989, 2021-04-30:
> @Lighty1989 : Yes with a lot of additions and some fix.

why don't you work together on the firmware? wouldn't it be better to share a github? I'll test yours tonight. It is important to me that its structure is like that of jyers, I like it very much.

Lighty1989, 2021-04-30:
@tititopher68 

I've already tested it, unfortunately it doesn't work as I did with the original firmware. i moved to my position across all axes and pressed set home offsets and that was my zero point until i clicked homeing again.

or what am I doing wrong?

the values ​​below are not zeroed either on the display X, Y and Z, they are fixed.

but the approach already what I want.

tititopher68, 2021-04-30:
@Lighty1989 
Unfortunately no, it can only work like this on Marlin bugfix 2.0.7.2 based firmwares. Unlike the Creality factory firmware version (1.0.2), the M428 command limits offsets to 20mm from the original 0 point or Endstops.
(cf:
 M428: Set home_offset based on the distance between the
 * current_position and the nearest "reference point."
 * If an axis is past center its endstop position
 * is the reference-point. Otherwise it uses 0. This allows
 * the Z offset to be set near the bed when using a max endstop.
 *
 * M428 can't be used more than 2cm away from 0 or an endstop.
 *
 * Use M206 to set these values ​​directly).

With my firmware version, the Offsets are well fixed. It is enough to make an M503 in a Pronterface or Octoprint console to verify it.
If you launch a Print, you will obviously see the axes make the home of origin for the Z axis, this is necessary to have a physical reference of the Z-Endstops, (it is basically a printer). And any G28 command will do it like this for the Z axis. On the other hand, the starting point of printing will be at the new coordinates defined by the Offsets.
for example then do a G0 X0 Y0 Z0, you will see the axes in their new position.
The configuration files (Configuration.h and Configuration_adv.h) will have to be modified to manage your laser head. If you configure one for this, then the autohome will be different. Also I invite you to configure files to manage a change of tools, The Hotend and the Laser head and then to integrate in your startGcode before the G28, the command for the change of tools.

tititopher68, 2021-04-30:
@Lighty1989 : 
New build V1.4.3.b.
I think this should now match your expectation as best as possible (with Marlin's limitations).

tititopher68, 2021-04-30:
@Lighty1989 : Sorry, but the file creality_dwin.h wasn't merged...Now it's OK