Jyers/Marlin #630: Probing failed on E3V2-UBL-BLTouch-10x10-HS-v4.2.2

ludvigaldrin, 2021-04-27:
### Description

Im running E3V2-UBL-BLTouch-10x10-HS-v4.2.2.bin on my Ender 3 V2 with Bltouch. Creating a Mesh works nicely however Autotilt Current Mesh is not working. I get Probing failed. This both on the machine direct in the meny or via octoprint. Running direct on a sdcard it still prints so cant see if it crashes or not.

Recv: Tilting mesh (1/3)
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: //action:notification Probing Failed
Recv: Error:Probing Failed

### Steps to Reproduce

1. In a build run after G28
2. G29 L1 then G29 J
3. Crash on octoprint  and from menu

**Expected behavior:** [What you expect to happen]

 I guess G29 J should probe 3 points and tillt mesh. But seems like it fails.

**Actual behavior:** [What actually happens]
It crashes. In ocroprint you get M112 and device shutdown. Looks to me that is trying to probe outside the print from a home position. I attach a video. 

https://user-images.githubusercontent.com/15027132/116294325-23a7d000-a798-11eb-884e-d5f719672571.mp4

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tome9111991, 2021-04-27:
You have set XY offset Probe?

ludvigaldrin, 2021-04-27:
I just got that tip from another person. So vill check that out. Probablly it. Also found this wiki page here: https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings with some tips.

Jyers, 2021-04-28:
If you haven't set your xy offsets that'll do it. How you managed to create a successful mesh without setting them is beyond me though...

gothcha, 2021-04-28:
In the start code you need to use G29 L0  the firmware stores the mesh in slot zero not slot 1 as used elsewhere.

ludvigaldrin, 2021-04-28:
@gothcha thanks. Changed to L0 now.
@jyers, done that now :). No crash anymore. I would say I have spent so much time leveling that the changes are probably so small right now that it works without the mesh. Thanks for your help and great firmware. I guess a tip would be to have a link from releases to Wiki. Found a lot good information there.

closed 2021-04-28 by ludvigaldrin

gothcha, 2021-04-28:
@ludvigaldrin Yes it can become a little obsessive, my V2 is the same. It worked perfectly well without the probe or the mesh, but as I bought it with the probe I felt I should use it!

Tuncay-Ayhan, 2021-04-28:
> @ludvigaldrin Yes it can become a little obsessive, my V2 is the same. It worked perfectly well without the probe or the mesh, but as I bought it with the probe I felt I should use it!

Obsessive? Nah xD. I think we all have that just a "little"