Jyers/Marlin #637: 10x10HS UBL Stops at Point 65

gmanc2, 2021-04-28:
### Description

I did not compile my self I simply downloaded the E3V2-UBL-BLTouch-10x10-HS-v4.2.2.bin for the 4.2.2 board and it ends the mesh at point 65 every time

### Steps to Reproduce

1.) attempt to level bed
2.) Stops at point 65
**Expected behavior:** [What you expect to happen]
It continues leveling 
**Actual behavior:** [What actually happens]
It stops at point 65
#### Additional Information

https://user-images.githubusercontent.com/4944080/116327843-9f544d80-a795-11eb-9882-3525c47e279c.mov

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-04-28:
See #419 and #403. I'm working on a status message that makes this more clear in the future.

closed 2021-04-28 by Jyers

gmanc2, 2021-04-28:
@Jyers Apologies, I looked and didn't see those two thanks for the response.