Jyers/Marlin #646: How to add correct ofset for fanduct and bltouch?

badplay, 2021-04-29:
I have a ender 3 v2 4.2.2 with newst firmware from this github

now i noticed that the newfanduct is a little bigger so the bltouch go obove the left side and try to lvl. i used https://www.thingiverse.com/thing:4647053 with the bltoch bracked v2

gothcha, 2021-04-29:
Set the XY probe to nozzle offsets in Control/Advanced menu. Typically -42 -9 this should then probe on the bed

aryajuanda, 2021-04-29:
just read the description on thingiverse :

"As I based my BLTouch mount on this one, the probe offset is the same (probe offset: X-42.00 Y-10.00).
Due to some people experiencing clearance issues with the first version of the bltouch bracket there is now a V2 version which has more clearance by moving the bltouch a little bit more to the side. The offset for the V2 bracket is X-43.65 Y-10."

gothcha, 2021-04-29:
Which firmware did you load BLTOUCH or UBL-BLTOUCH?
So when you add the correct offsets what exactly happens when you run the bed levelling?

closed 2021-04-30 by Jyers