Jyers/Marlin #658: Manual tuning microstepping issue

TheNeoShogun, 2021-05-01:
Firmware version 1.3.1 UBL 15x15 BLTouch 4.2.2. 
After probing UBL with BLTouch mesh 15x15 there are points that are off on Z axis. With Manual Tuning I tried to set proper Point Z offset and using nob left/right to set value works but microstepping(up and down) makes Z offset behave like it "cached" the value and going to next point sets nozzle on the height of previously set point using microstepping up/down and going to other points keeps the nozzle on "cached" value.
To reset this "cached" value I had to go to "Prepare -> Z offset -> Home Z offset" to reset the value. 
I have tested it by putting manually one value way off and made video that is under this link
https://youtu.be/IdJZh6M5c74

tititopher68, 2021-05-01:
@TheNeoShogun 
I just tested according to your video with my build and Jyers's last build and i have no issue.
Have you compiled your firmware with your own configuration files?
Note that you are not currently using the latest firmware version. Indeed, the item NEXT no longer exists, that may have been your issue...

TheNeoShogun, 2021-05-01:
I use firmware from release section, I didn't compile any code.

Jyers, 2021-05-01:
Just published a new release with a revamp of the leveling menu. Let me know if the new version experience the same issue.

TheNeoShogun, 2021-05-02:
It's fixed but taking away "Next point" feature significantly reduced efficiency on quick mesh check-up a specially on  15x15 grid. One click vs click, turn by 1 value, click :-(

Jyers, 2021-05-02:
Hmmm, I guess I never expected someone would want to go through the whole mesh like that. Makes sense though, I'll probably add it back then, sorry for the inconvenience.

closed 2021-05-02 by Jyers

TheNeoShogun, 2021-05-02:
Going through whole mesh with "Next point" is quite fast, I haven't measure it but subjectively 1s per point.
Thanks, I appreciate it. You are doing a great job, the firmware is awesome.

tititopher68, 2021-05-04:
@TheNeoShogun : code for ->
[image]
[image]