Jyers/Marlin #665: E3V2-BLTouch-5x5-v4.2.2 uses UBL instead

leandrotoledo, 2021-05-02:
I just downloaded and updated the firmware on my Ender 3 V2 w/ BLTouch from v1.3.1 -> v1.3.2 and I noticed that under the Level Menu it's offering UBL-related functions e.g. Create New Mesh, Mesh Viewer etc.

Steps to reproduce:
* Downloaded bin https://github.com/Jyers/Marlin/releases/download/v1.3.2/E3V2-BLTouch-5x5-v4.2.2.bin
* Renamed firmware in the SD card from firmware6.bin to old_firmware6.bin
* Copied E3V2-BLTouch-5x5-v4.2.2.bin into SD card as firmware7.bin
* Powered on the printer
* Hit Level on the main menu, the header says "Unified Bed Leveling"

See photos attached:

[image]
[image]

Maybe I missed something?

One suggestion, maybe under Info it could show the sub-version of the firmware as well? Like, under Build Number, "E3V2-BLTouch-5x5-v4.2.2" etc. Thoughts?

Thanks for the excellent firmware!

darco415, 2021-05-02:
sd card should always be blank before adding new fw, and yes renaming from previous fw is a must

leandrotoledo, 2021-05-02:
> sd card should always be blank before adding new fw, and yes renaming from previous fw is a must

I don't think the first statement is true. I successfully reverted back to v1.3.1 (w/ ABL) simply iterating the corresponding bin as firmware8.bin in the same (not blank) SD card.

Jyers, 2021-05-02:
This is simply a label mistake. As said in the release notes, all leveling versions now have a leveling menu. I just forgot to change the menu title from Unified to just Leveling. What you have is bilinear, not UBL

closed 2021-05-02 by Jyers

Jyers, 2021-05-02:
FYI, this was a small enough fix and close enough to when I published the release, that I have simply re-done the release with the fix.