Jyers/Marlin #666: Cooldown weirdness at the end of file.

rakosi2, 2021-05-02:
### Description

I have an [auto-power off switch](https://www.thingiverse.com/thing:4779435). During cool down, you can go into the menu and it would go past my set temp.

### Steps to Reproduce
I have added the following to the end of my code:
```
M140 S0 ;Turn-off bed
M109 R50 ;Turn-off hotend and wait

G1 X0 Y229 ;Turn off printer
```

**Expected behavior:** I expected it to continue printing so I still see the pause stop buttons.

**Actual behavior:** I get the Confirm button can interact in the menu. It will cool down to 27 degrees then proceed to shutdown.

#### Additional Information

I have tried Marlin 2.0.8 and it cooled down as expected. 
I use the precompiled E3V2-UBL-BLTouch-15x15-HS-v4.2.2.bin and it happens for both 1.3.1 and 1.3.2
[test gcode.txt](https://github.com/Jyers/Marlin/files/6410713/test.txt)
[image]
[image]

tititopher68, 2021-05-02:
@rakosi2 
On the precompiled firmware and in the configuration's files Emergency Parser is actived. I think that's what explains this behavior.  Try commenting out // #define EMERGENCY_PARSER in the Configuration_adv.h.  and recompile ...

rakosi2, 2021-05-02:
Same thing. I commented out //#define EMERGENCY_PARSER and it still allows me to go to the menu and it passes set 50c to 24c

Jyers, 2021-05-02:
I believe this is a result of `#define PRINTJOB_TIMER_AUTOSTART` Try disabling this option in configuration.h

rakosi2, 2021-05-02:
That did seem to fix it. It did start beeping(End G-code) during the purge line(testing with only a purge line) but that could just be Marlin looking ahead.

Jyers, 2021-05-02:
Yeah, that would just be it finishing the commands early. Glad that did the trick!

closed 2021-05-02 by Jyers