Jyers/Marlin #669: Set Home Offsets

Lighty1989, 2021-05-02:
hello jyers, i wanted to ask again, did you write that down somewhere with the set home offsets on all axis? that shouldn't be a big act for you. 

maybe you can also create a new submenu prepare> laser settings> set home offsets or where you can move the axes, that you can set the zero point right there ...

I don't care, the main thing is that it would be a shame if I constantly have to flash firmware just to be able to laser.

I check your github every hour in the hope that there is something new.

tititopher68, 2021-05-02:
@Lighty1989 
Did you only test the very last version that I offered to you? I remind you that the factory version functionality is no longer relevant at Marlin 2.0.x ... You can check it on the Marlin site concerning the Set home Offset  (M428 and M206).
 I invite you to test what is offered and if it is sufficiently satisfactory, I would put it in Pool Request.  @Jyers will integrate it or not (I think this is not a priority).  But in the meantime you will be able to integrate it  for yourself ...

Lighty1989, 2021-05-02:
Oh okay, i test your last change.

Jyers, 2021-05-02:
I think I have a way to implement this almost identically to how the stock firmware handles it. I'll work on it today

Lighty1989, 2021-05-02:
@tititopher68 

Unfortunately, it is not the desired effect.

Normally it should be that you approach all points via Auto Home. Then you put your axis on the position you want.

If all axes are in the desired position, Set Home Offset should take the position as the zero point.

Unfortunately this does not work with your changes, as soon as I load a file it starts again at the regular zero point of Auto Home.

There is still a mistake somewhere.

It is best if you take the original firmware and look at what the firmware is doing

Jyers, 2021-05-02:
@Lighty1989 Just implemented the option. In the motion menu there is now a set zero position option. It won't change the on screen coordinates, but the workspace offsets will make it so whatever position the head it at is considered zero when running your gcode. Can you compile your own firmware or would you like me to send you a compiled file so you don't have to wait for the next release?

Lighty1989, 2021-05-02:
@Jyers thank you, i test it now and give feedback

tititopher68, 2021-05-02:
@Jyers : I have already offered this to him and it was not good ... For him, the autohome must do this at the new coordinates (????). What I did, set the offsets to the chosen X, Y, Z values and then if we launch a Print, the 0,0,0 point is well shifted to the Offsets values, including the autohome. But I absolutely do not understand what he wants.
If you do a G92 X0 Y0 Z0, after positioning the axes, the autohome returns to the starting point 0,0,0, not to the one that would have been defined ....
I am doing a video for that....

Jyers, 2021-05-02:
@tititopher68 Yes, but I'm assuming he is not auto-homing. For a laser gcode you don't auto-home, instead you "manually home the printer" by zero-ing all the axis like I implemented with G92.

Lighty1989, 2021-05-02:
Apparently there is a misunderstanding here

I have now tested the Jyers and it works like the original firmware

It may well be possible that it will work for you too, but I don't know how.

but i don't know the coding either, i can only say that it works from jyers.

I can move axes where I want under move, then I click on Set Zero Offset and go straight to Print and it lasers where I have defined it.

Great, I'm happy for now. To round things off, a beep tone after clicking on Set Zero Offset would be nice, and if it also sets the values ​​at the bottom of the display to zero.

Maybe Jyers can answer you again. I also don't speak English very well and I use a translator

Lighty1989, 2021-05-02:
> 
> 
> @tititopher68 Yes, but I'm assuming he is not auto-homing. For a laser gcode you don't auto-home, instead you "manually home the printer" by zero-ing all the axis like I implemented with G92. If you've ever used a cnc, you know that you don't home to the endstops, you home to the work piece so the machine knows where to put the pattern.

Exactly, that's the way it is, CNC does exactly as you describe it.

tititopher68, 2021-05-02:
I know all that but.... @Lighty1989 says : "Unfortunately this does not work with your changes, as soon as I load a file it starts again at the regular zero point of Auto Home.". 
It is therefore necessary that he removes from his Gcode the G28 and all G92.1 or else he remakes his Gcode with the choice "No Homing" ....

Lighty1989, 2021-05-02:
[image]

This is what the gcode looks like from a laser file

possibly that clears up something

Jyers, 2021-05-02:
> I know all that but.... @Lighty1989 says : "Unfortunately this does not work with your changes, as soon as I load a file it starts again at the regular zero point of Auto Home.".
> It is therefore [...]

I don't think he meant he was auto-homing when printing as that would defeat the point of this whole thing. I think he just meant it went to 0,0 as in the last time he homed.

closed 2021-05-02 by Jyers

tititopher68, 2021-05-02:
it's exactly what i propose.
if I run a Print with the Offset defined, my point 0,0,0 is well offset by the value of the offsets ....
In the video :
# 1 Without Set Home Offset: I run a print which contains (for the example a G28, but which first prints 2 purge lines at 0,0,0.)
# 2 I set in my example the offset to 50,50,50. The autohome at the end of the cycle is positioned at 50,50,50
I restart the same print and the purge lines are at 50.50.50! This means that even if you then move your axes manually, and restart the print, the first point will still be at 50.50.50 and this, as long as you do not do a Reset Home Offset ...

tititopher68, 2021-05-02:
video : https://e.pcloud.link/publink/show?code=XZQC6XZeUp8EykxFTh97AiprNlQ0RetvGxy

Jyers, 2021-05-02:
But there's no reason to have persistent offsets. I'd imagine it's easier for the user if they do reset themselves when homing, that way you don't have to worry about the coordinate space being offset for any other functions. I see what you're trying to accomplish with your method, but I don't think that's what the goal is here. Simply setting the position to 0 0 0 is all that is needed.

Lighty1989, 2021-05-02:
Now I understand your thinking but it's easier as jyers says. i would like to print mostly and for lasers every now and then that is completely sufficient and after the limit switches are approached again it is back to the basic position for printing. nevertheless many thanks for your effort.

Many thanks for your help @tititopher68 @Jyers 

but the way I wanted it and know it from creality firmware, jyers implemented it that way.

Lighty1989, 2021-05-03:
Hello jyers, your change works great. Can you Move set zero offset Button to Move Menü under Move X, Move Y, Move Z?

This is my last wish.

Jyers, 2021-05-03:
@Lighty1989 You're right, that's probably the better place for it

Lighty1989, 2021-05-04:
@Jyers everything can be set quickly under move without having to go back to another submenu, what can I do to move it there for myself?

tititopher68, 2021-05-04:
@Lighty1989 You are really impatient ;-)
[image]
[image]

Lighty1989, 2021-05-04:
Okay, i test it. Thanks for your help @tititopher68

Lighty1989, 2021-05-04:
@tititopher68 it works :-) thank you
@Jyers this icon in white/blue... more cnc like ^^
[image]

Lighty1989, 2021-05-06:
> 
> 
> Obviously if I put a part below, and that I do before my autohome with this part, I leave the Home z Offset at 0...and finally it does not matter then the Gcode with or without G28
> 
> video : https://e.pcloud.link/publink/show?code=XZQC6XZeUp8EykxFTh97AiprNlQ0RetvGxy

@tititopher68 who i can find the dwin_set you are using on your display (see your video) i like this icons

tititopher68, 2021-05-06:
@Lighty1989 
In my build : https://github.com/tititopher68/Marlin/tree/Ender_3_V2_Extensible_UI/Display%20firmware/Firmware%20NEWS3%20Icons/DWIN_SET