Jyers/Marlin #672: Newest 10x10 has update

jwkopp, 2021-05-02:
During auto bed level process is it normal for the probing to not finish?  What I mean by is that it probes about 70-75 points then stops and tell me it has probed as many spots that could be reached.  Is that normal for auto bed leveling?  Or is something off.  Looks like the outside edge of where it probes is at the 220mm mark.  Should I change my bed size to 235 to get all the points?

Jyers, 2021-05-02:
No, that's correct. Funnily enough I added that "Probed all reachable points" popup to try to mitigate this question (I get it a lot) Maybe I should add "Everything is fine I swear" to the save level popup too... Anyway, yeah that's just how UBL handles the mesh, all other points will be inferred.

closed 2021-05-02 by Jyers

jwkopp, 2021-05-02:
Ok I wasn't worried about the prompt just making sure I was doing everything correct