Jyers/Marlin #678: Kick-start parts cooling fan

kp-bit, 2021-05-03:
Would it be possible to implement a configurable threshold for when the parts cooling fan needs a kick to get going?

The problem is that when you have a 12V fan and buck converter, the fan is not handling PWM well at low rpm's. So when the parts cooling fan starts up at speed 85 on layer 2 (or whenever), it just rocks back and forth. A gentle nudge will get it going. Usually it needs to get over 200 to start itself.

So if a configurable value was possible, that value could be sent to the fan for just a second to get it going if the gcode value is lower, before dialing down to whatever value is in the gcode and continuing. I don't know if that extra short blow of air would be problematic for the printed item though.

Thanks for the great work you do!

Jyers, 2021-05-03:
There is a configuration option for this in case you're able to compile your own firmware. If not I'll look into if it would be possible to expose this setting on the screen.

vhristenco, 2021-05-03:
Set fan_min_pwm to the lowest value that makes your fan spin.

kp-bit, 2021-05-03:
@vhristenco, wouldn't that make the fan blow at 200 when only for example 85 is wanted?

Jyers, 2021-05-03:
@kp-bit Yeah, what you want is `FAN_KICKSTART_TIME`

gothcha, 2021-05-03:
The buck converter is not at all ideal under PWM, the PWM is basically reducing the average drive voltage. The buck converter will be trying to maintain a fixed 12V output for all input voltages.

kp-bit, 2021-05-03:
@Jyers, awesome if you can somehow expose that in the menu! If not I guess I'll have to start compiling myself...

Jyers, 2021-05-03:
I'll put it on the todo list 👍

closed 2021-05-03 by Jyers

kp-bit, 2021-08-14:
Just a gentle nudge, hoping this will make it into a future version... :)

Jyers, 2021-08-15:
Unfortunately it looks like it's a compile time only setting 😢 

If you want any help getting started with compiling feel free to let me know! It's not as hard as it seems I swear 😄

kp-bit, 2021-08-29:
Aww, that's a shame but thanks for looking into it! I have tried compiling, it's just so much easier picking up your latest build... :)