Jyers/Marlin #696: Download only allows Manual Bed level with my bltouch

BulldawgDesigns, 2021-05-04:
I dowloaded and flashed my Ender 3 V2 with E3V2-BLTouch-5x5-HS-v4.2.2 and it will not regocnize my bltouch.
When I go to the level prompt it only allows me to to create a manual level.  I cannot auto level.
I have tried several re flashes and have the same issue.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68, 2021-05-04:
I think it's a material BLTouch issue...

BulldawgDesigns, 2021-05-04:
What does that mean?
I can use an older version and it works.  The new version doesn’t.

Jyers, 2021-05-05:
Just to confirm, if you go to the leveling menu and select "Create new mesh" it goes to a manual leveling screen and does not start leveling automatically?

tititopher68, 2021-05-05:
@tititopher68 , do you have the Probe X Offset item or Probe menu into Menu Advanced...If Not, it's a matérial issue.
The firmware test if you have a bed Probe (HAS_BED_PROBE), if not , you get the manual Mesh menu...

tititopher68, 2021-05-05:
I am correcting my previous post, because it is imprecise.
In the firmware there is a HAS_BED_PROBE test. This parameter is at 1, if there is a BLTouch for example in the Configuration.h file (#define bltouch).
When we compile the firmware, it is first checked (according to Configuration.h) if we have chosen the UBL mode. If this is the case, is tested if we have a leveling probe, etc ... If we have not chosen the UBL mode but Bilinear (Auto_Bed_Bilenaire_Leveling), HAS_BED_PROBE is tested, if the value 1 is returned, then it launches the G29 expected, otherwise, it offers the Manual Leveling menu.
So either there is an error in the precompiled files (#define bltouch was commented out either // # define bltouch), or the BLTouch is faulty. Mesh cannot be done and you arrive at the possibility of Manual Leveling.

We can quickly do a check, if you have the Probe menu or the possibility of having the Probe X Offset setting for example. This means that #define bltouch has been activated in Configuration.h and therefore it is a problem with BLTouch. If these functions do not appear then it is about a Configuration.h error in the precompiled files ...

closed 2021-05-06 by Jyers