Jyers/Marlin #697: Manual UBL Bed Leveling z offset doesn't change or jumps.

arballz, 2021-05-04:
### Description

When manually leveling the bed using UBL on the UBL-Noprobe firmware the number sometimes doesn't change or jumps over a number

### Steps to Reproduce

1. go to create new mesh and click on microstep down.
2. When you reach -0.09 you will push microstep down and it wont change.
3. If you push it again it will change to -0.10 then -0.11 then push one more time and it changes to -0.13

**Expected behavior:** to change by 0.01 every time i push the button.

**Actual behavior:** at -0.09 when the button is pushed nothing happens. at -0.11 when the button is pushed it moves to -0.13 not -0.12. Going back up it doesn't have the same issue.

tititopher68, 2021-05-04:
Well no, because you forget that in reality, the precision of the point is less than a hundredth of a millimeter and therefore, to make the display visible, the value is rounded ... so for example 0.01 can be 0.014 or 0.009 in reality...

arballz, 2021-05-04:
I was wondering if that was the case but didn't have the data to confirm. If a rounding issue is the problem then I will close this as long as the actual value is recorded and not the rounded display value.

Jyers, 2021-05-05:
Yes, I can confirm that this is simply a rounding issue, I may change how the display values are rounded in the future to avoid confusion.

closed 2021-05-05 by Jyers