Jyers/Marlin #715: [BLTouch] Create new mesh vs Autotilt current mesh

TheNeoShogun, 2021-05-06:
I have firmware 4.2.7 UBL 15X15 BLTOUCH v1.3.2
I noticed that with warped glass bed it's significant difference on how autotilt and create mesh is measuring with BLTouch.
First image is create mesh and checked with manual tuning that nozzle touches the glass bed in the middle and on the sides. 
[image]
Second image is autotilt that shifts whole mesh upwards. 
[image]

I think this issue is caused by different starting point for measurement. When it's create mesh starting point is in the middle of the glass bed. When it's autotilt starring point is bottom left corner and autotilt assumes it's the reference point(Z height).

I noticed it when after create mesh I started printing and starting G-code loded the mesh and 4 point autotilt caused nozzle to scratch the glass bed in the middle.

gothcha, 2021-05-06:
Is the variation truly almost 3mm? If so I assume individual values cannot be displayed above 0.99mm? I would get a decent bed, mine is within +\- 0.2mm. The tilt works perfectly (3 or 4 points) for me.

TheNeoShogun, 2021-05-06:
The BLTouch supposed to fix uneven glass bed so you wouldn't have to buy new one. If you have flat even glass bed then BLTouch is unnecessary.
Other problem with buying new glass bed is it can also be uneven and you could ended up buy for example 10 glass beds and all uneven and warped.
I bought BLTouch because I couldn't get print after properly levelled glass bed.
Buying new glass bed is not a fix to this problem it's just skipping it, if you are lucky to stumble upon even glass bed. 
The problem is that autotilt has different point of origin for measurement then create mesh and takes it as reference point. If your glass bed if perfectly flat then you won't notice this bug. The point for reference for autotilt and create mesh should be the same point. And it's not so it's a bug.

gothcha, 2021-05-06:
Sorry I was trying to understand the variations from the view mesh screen. There appear to be adjacent mesh points that go from almost zero to max green (front right 3-4) Almost like there is a step? Or are the colours not displaying correctly?
The autotilt will be homing at the centre (ref origin) then measuring a selection of points and comparing the values to the current mesh. The order of tilt points is not defining the origin.  If nothing has changed the mesh should stay the same. The issue could be that if the tilt probe points are not at an identical point compared to an original mesh point then the interpolation error could confuse it? If there are large local variations near the tilt points this will exaggerate the errors. The tilt points by default are at the extremes.  If you self compile then you can specify the points and avoid odd areas of the bed. Maybe other users have further info?

gothcha, 2021-05-06:
The tilt routine is defined/controlled in the standard Marlin and Jyers firmware does not impact on its operation simply runs the specified G codes.

TheNeoShogun, 2021-05-06:
If I could code and compile I would probably work on it myself but I don't.
If both autotilt and create mesh home on the same red point, why such a huge difference? The zero point for both should be the same value if it's starting in the same origin, right? Then the variation should be minimal. Even using 64 points autotilt the diffrence shouldn't be that big but it is.
If tilt is not coverd by the Jyers firmware then meybe create mesh has a bug.

gothcha, 2021-05-06:
@TheNeoShogun  - can you confirm that the bed variation is actually 3mm and if there are steps/lumps in the corners? The auto level sadly cannot perform miracles, it will reduce the variation but the remaining error is a function of the original error, and rapid variations are a significant limitation to the effectiveness. If you remove the glass itself is the metal plate flat? I was just surprised that on a V2 with the new glass beds that you could have such a huge variation. Maybe I was lucky?
I know some people who just use a piece of flat glass for the bed.

gothcha, 2021-05-06:
Also you do not have to run the Autotilt (G29 J) in start code, if you are happy with the stored mesh result remove the G29 J and just leave the G29 A and G29 L0 to activate and load the mesh. This may avoid the issue and save time?

TheNeoShogun, 2021-05-06:
@gothcha I'll check the variations later. The glass bed was clean before create mesh and autotilt. Not looking on one specific point but on overall view, you can notice that its like whole bed shifted on Z axis on all points.
If there is like few points off with autotilt using 64 points, shouldn't there be plane tilted little bit instead of shifted whole thing ?
I'm just trying to understand how this thing work, I don' t know the details but logically reasoning it should't go like it does right now with my printer.
May be there is some difference in using Z offset value in the firmware, the one from Prepare->Z offset menu and one from homing in autotilt and create mesh function.
I think that when create mesh is done and then autotilt is done right after without doing something in between it should basically show the same thing if nothing changes.
It may happen that after the test print I may press the bed or hit it while taking off the print so I assume that I will need do autotilt before next print to adjust to minor changes due to that.
I also noticed that autotilt can measure the right edge of glass bed with X at 245 mm and I think that create mesh doesn't go that far but I guess it's different issue.

gothcha, 2021-05-06:
No I think that may exactly be the issue. I was concerned about exactly this type of thing so I compiled with the mesh and probe offsets set such that all 100 points are measured and that the 3 tilt points match the mesh points. This way there is no interpolation/extension of the mesh. This has worked for me. If is very easy to compile your own, I was a little nervous initially but found it straightforward using Visual Studio Code.

TheNeoShogun, 2021-05-06:
@gothcha Just noticed that Autotilt in Level menu doesn't home it just go to left bottom corner and start to measure.
The create mesh does home. I use 15x15 UBL v1.3.2 nad mainboard 4.2.7 and autotilt with 64 points(grid 8).
I experimented with Z offset from Prepare menu and Autotilt from Level menu. 
I put the Z offset to -2.50 and run Autotilt then I change Z offset to 0.00 and run Autotilt. 
The correctnes of real Z offset is irrelavent to this experiment.
First -2.50

[image]

Then Z offset 0.00
[image]

Also I made another autotilt from level menu after last tilt. Z offset 0.00
[image]

Then Prepare menu -> Move -> Z axis -> 20.0 , Z offset still 0.00
[image]

I may found something intresting.

TheNeoShogun, 2021-05-06:
@gothcha autotilt is not always homing, now put Z axis i Move to 30.0 and it did home but I it was idle for a while so I have to check if it's also time based thing.

gothcha, 2021-05-06:
Just tested mine and after power up and load mesh the Autotilt homes first as it should. If it has already homed then it will not rehome on a second tilt.

TheNeoShogun, 2021-05-06:
@gothcha it is possible that someone can physically by accident move printhead, put arm on X gantry or move the bed and it's not showing on coordinates on screen so to prevent possible errors I would put home before any operation that changes the position of the nozzle.

gothcha, 2021-05-06:
No the Z offset will have no effect on the BLTouch or levelling, just impacts the nozzle height during print. 

The default builds have mesh inset at 10 and probe margin at 20, so the marlin code will have to interpolate/guess the outer 10mm of the mesh which it cannot probe. Plus of course the final 10mm outside the mesh is not calculated at all. So 20mm around the edges is a little suspect. If your bed is pretty flat this is not a big issue but fir you it probably is.

gothcha, 2021-05-06:
A valid point, something for @Jyers to consider? But more than likely unrelated to your issues.

gothcha, 2021-05-06:
It homes again if the steppers have been disabled.

TheNeoShogun, 2021-05-06:
Using the same mesh without loading for few times.
Another tries: Move -> Z 30 - it homed.
[image]

Then right after autotilt another autotilt the Z is 10.0 after previous one. No home.
[image]

Should Autotilt give the same results when done one after another ? No changes were made, in theory I think autotilt should give the same measurements.

Then Level Menu -> Load Mesh and autotilt. Also no home.
[image]

Then Prepare -> auto home, -> Level Menu load mesh and autotilt
[image]

I'm really confused how Autotilt should work.
As I understand I do Create Mesh then if something changed like accidental bed screw push, disable steppers or pressing by accident the glass bed I should do Autotilt to correct the mesh by thing that changed their position to one measured.
If nothing changed then autotilt should give the same measurement done twice in a row or the same after loading the mesh and autotilt.

gothcha, 2021-05-06:
Have you run the BLTouch repeatability test from the Control/Probe Menu? This will give the variation between probes to verify the BLTouch is working correctly. I does this at the centre of the bed. If this is OK then other than what I have already indicated I cannot shed any more light on this sorry. You could try changing the tilt mesh setting to 1 (currently 2)in the UBL settings menu, this will use a three point tilt mesh and see if this is any different. It will use the centre back instead of the back corners.

TheNeoShogun, 2021-05-06:
Now, Level menu -> load mesh, Prepare menu -> disable stepper ,  Level menu -> autotilt.
[image]

Ok, I'll try run probe test.

TheNeoShogun, 2021-05-06:
Deviation 0.026327.
What does it tell me?

TheNeoShogun, 2021-05-06:
Ok, deviation changes every test.
0.122139(Count 5) , 0.006209(Count 4) , 0.006289(Count 4), 0.006436(Count 4), 0.004204(Count 4)

gothcha, 2021-05-06:
The 0.006 numbers look fine but the 0.12 looks odd - I believe this is the spread in mm
Should be less than 0.01mm I would guess. 
Perhaps there is an issue with the BLTOUCH?

gothcha, 2021-05-06:
Are you running the mesh generator with the bed at operating temperature? Same question for the tilt. It is advisable as the bed can warp over temp. This is not related to your inconsistency however.

TheNeoShogun, 2021-05-06:
I can't tell if anything is wrong with BLTouch, looks ok.
Bed temp is 55 C while all the measurements.

gothcha, 2021-05-06:
Great stuff on the temp. What Probe XY offsets are you using. Maybe I can compile a version for you with a fully probable 10x10 mesh and see if that helps. However if there is an issue with the repeatability then you will still have problems. Run the test with 20 points or so a few times and see what this gives. There have been issues with the BLTouch causing probing failures. See thread #192 - bottom line is if you are using the 5 pin BLT connector there may be a need to add an extra pull-up resistor or use the Z_Min port and self compile firmware!!  Happy days!!

TheNeoShogun, 2021-05-06:
@gothcha just switched to 10x10 firmware and autotikt grid 2 and looks better. Less differences. 
Create mesh
[image]
Autotilt
[image]

gothcha, 2021-05-06:
Fabulous this looks perfect. I must say I have only ever used the 10x10 mesh. Hopefully you can now print away to your hearts content :) Your bed is now within +/- 0.2mm as I would have expected.

Would be interesting to see if anyone else is having any issues using a 15x15 mesh?

TheNeoShogun, 2021-05-06:
@gothcha I'm using 5pin BLTouch port on Creality Marlin 4.2.7. Autotilt on already tilted mesh still having problems.
[image]

At least it's fairly accurate on first run.

Yes, I'm interested on how other people have issues. 
I'm now suspecting that autotilt calculations may have small error somewhere which amplifies with running repeatedly autotilt.

TheNeoShogun, 2021-05-06:
@gothcha this is insane. I lubed the probe with ceramic grease and I got more consistent results. 
Results of M48 vary in range of 0.0005.. to 0.003...

adam3654, 2021-05-06:
> @gothcha this is insane. I lubed the probe with ceramic grease and I got more consistent results. 
> Results of M48 vary in range of 0.0005.. to 0.003...

What is the name of the ceramic grease you used? I've had the consistency problem in the past.

TheNeoShogun, 2021-05-06:
@adam3654 CX80, I put a little bit and wiped off excess grease so theres very thin layer.

adam3654, 2021-05-06:
> @adam3654 CX80, I put a little bit and wiped off excess grease so theres very thin layer.

Thanks @TheNeoShogun I'll try to find some for future use.

Jyers, 2021-05-06:
@TheNeoShogun Have you tried a larger probe grid? The grid size can be set in the UBL settings menu. A 1=3 point, 2=2x2,etc
A 2x2 would definitely be more accurate, and a 3x3 has a probe point in the middle, so maybe that would prevent it from getting offset.

gothcha, 2021-05-06:
@TheNeoShogun Good idea on the lubrication, I have seen this done before.

TheNeoShogun, 2021-05-06:
@Jyers here is comparison. 
Create Mesh 10x10.
[image]
And load mesh, autotilt from 1 do 8 grid.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jyers, 2021-05-06:
Very interesting, it seems larger meshed move the whole mesh up. Might be due to how they reference the mesh

TheNeoShogun, 2021-05-06:
@Jyers this is load mesh, autotilt, change grid size repeat(no auto home from tilt or manualy from prepare). Now I'm checking load mesh, auto home from prepare, autotilt, change grid repeat from 1 to 8. And I think this will also be interesting.

TheNeoShogun, 2021-05-06:
@Jyers now that's interesting. 
The same mesh 10x10 
[image]

Reverse order(just for fun) grid from 8 to 1. With steps load mesh, Prepare -> auto home, level -> autotilt,  change grid,  repeat.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jyers, 2021-05-06:
I'd say that's pretty conclusive. Autotilt should always home first. I'll make it happen 👍

closed 2021-05-06 by Jyers

TheNeoShogun, 2021-05-06:
@Jyers great, thanks!