Jyers/Marlin #716: Micro Swiss Direct Drive

DominicCasper, 2021-05-06:
The Micro Swiss Direct Drive doesnt work with your Firmware. 

I must set:
#define X_MIN_POS -5
#define Y_MIN_POS -10
on the Configuration.h.
 Is it possible to recompile or modifiy the .bin file?

Thanks
Dominic

smooreace, 2021-05-06:
It works perfectly, Especially since the Firmware has zero direct function
involving anything outside the screen and controller. But you have to
compile your own firmware, as instructions tell you.

DominicCasper, 2021-05-06:
@smooreace thanks for your help.
Can you explain me how to compile my own firmware?

smooreace, 2021-05-06:
Nope... there are probably hundreds of guides and videos out there
pertaining to this. No point in reinventing the wheel

DominicCasper, 2021-05-06:
@smooreace ok thanks :)

closed 2021-05-06 by DominicCasper