Jyers/Marlin #717: Issue with new pack icons (include Gotcha icons and AxisBL, etc...)

tititopher68, 2021-05-06:
( #define CREALITY_DWIN_EXTUI_CUSTOM_ICONS actived)

With the Jyers icon pack (including the new AxisBL icons, etc.) there is a bug.
In principle, the screen firmware only accepts a limited size for the 9.ICO file. And there, if you go to menu Prepare -> Manual Level, the menu is not displayed completely and hangs!

For my part I deleted the 4 Chinese icons entries (# 82, # 84, # 86, # 88) in 9.ICO and in DWIN_ICO.py and more bugs!

Jyers, 2021-05-06:
I need to reorganize the icon packs and sort some stuff out anyway. I'll update this when I've got that done. I think you need the MLevel icon pack though, as I don't know if the others have the icons for manual leveling, that's probably your issue.

closed 2021-05-06 by Jyers

tome9111991, 2021-05-06:
I dont have issuse with Mlevel icon pack 🤔

tititopher68, 2021-05-06:
@tome9111991 but i don't know if you have all the news icons...@gotcha have the same issue
You have too this issue...You wrote : "Small Bug: Center Icons looks like not centered i dont know why."
Look at this discussions : https://github.com/Jyers/Marlin/issues/562

tome9111991, 2021-05-06:
> @tome9111991 but i don't know if you have all the news icons...@gotcha have the same issue
> You have too this issue...You wrote : "Small Bug: Center Icons looks like not centered i dont know why."
> Look at this discussions : #562

Okay true.
I dont use the new icons

mgnthe54, 2021-05-25:
Hello I'm french.
I hope my English is not too bad!
You did a great job, thanks to you the Ender 3V2 can finally fully give its capabilities!

I have a suggestion is it possible to display in real time which layer is being printed?
 (if we took a break or something else we would know if we are far from it)
Thank you

PS: do you think one day develop a French version of your Firmware? (thank you Google Translate...)

Jyers, 2021-05-25:
@mgnthe54 Hi! 
Displaying the current layer can be achieved through the octoprint plugin DisplayLayerProgress, but isn't really possible with sd printing.
I would love to have multiple languages available in the firmware at some point, so stay tuned!

mgnthe54, 2021-05-25:
Thank you for his information!
Too bad for the layer from the SD ...
Thank you again good continuation!

tititopher68-dev, 2021-05-26:
@mgnthe54 
Bonjour, non c'est tout à fait possible en ajoutant au fichier Gcode des M117...Je peux vous mettre en lien un fichier python (très basique) qui ajoutera à chaque changement de couche dans le Gcode de votre fichier sur votre SD une commande M117 Layer X/nbre total de couche.
Lors de l'impression, s'affichera à chaque changement de couche le message sur l'écran, comme avec Octoprint.
Le fichier fonctionne avec des Gcodes venant de Cura ou PrusaSlicer, mais je peux l'adapter pour d'autres sans pb.

mgnthe54, 2021-05-27:
Bonjour titotopher68 (je suppose Christophe et Alscacien?) moi c’est Gilles et 54 pour la Meurthe et Moselle... Je suis débute dans le monde du 3D et je ne suis plus tout jeune...

Super boulot que vous avez effectués, comme je le disais cela permet à la Ender 3V2 de réellement être beaucoup plus performante!

J’utilise Cura pour découper mes fichiers STL.

Je veux bien le lien du programme Python pour rajouter sur les fichiers gcode l’affichage de la ligne actuelle sur la totalité des lignes de la pièce...
Je suppose que c’est un post traitement à appliquer après Cura (ou autre) sur les fichiers gcode. (on ne peut pas paramétrer directement Cura pour faire cette tache?)

Effectivement la fonction M117 affiche un message à l’écran LCD d’après https://marlinfw.org/docs/gcode/M117.html

L’affichage du layer se fera de façon fugitive sur l’écran ou indiquera en permanence le layer en cours? Ce n’est pas indiqué dans cette page web...
J’ai rentré manuellement dans un fichier gcode un  :
M300  ; beep
M117 Layer 0 ; texte        et le Layer 0 s’affiche bien et reste présent à l’écran tant qu’on ne fait pas au moins un M117; sans rien pour efface le message

Si je peux un peu abuser dans les différents tuto Anglais d’où mes difficultés (mes 2 ans d’Anglais scolaires date de 45 ans...) on voit ajouter dans le Cura après le G28 qui remet la tête ne position de départ :

soit G29        un nivellement du lit: linéaire, en 3 points,...

soit M420 S1        active la mise à niveau! Enfin je sais pas trop ce que cela fait réellement, je suppose que l’on utilise la carte des niveaux que l’on a stockée depuis level –>Creat new Mesh (et leveling Active coché utile ou pas?) et qu’il y a une correction du zoffset en temps réel en fonction de la position de l’extruder?

Quel est le mieux?

Merci d’avance

PS: une version Française du firmware est-elle un jour prévue et/ou une notice Française (à défaut en Anglais;;; avec google traduction et quelques essais pratiques) de ce que font réellement les fonctions de votre firmware pour les débutants comme moi?

Merci encore

REPS: je suis équipé d’un Ender 3V2 carte mère 4.27 Dual Guard, BLTouch Creality et des derniers firmware 1.3.3 (25 points lent) et l”écran le custom

tititopher68, 2021-05-27:
Suite à une remarque de Jyers sur ce genre de discussion, je vous invite à me contacter via l’adresse suivante : ***@***.*** Je vous donnerai l’accès à une version du firmware plus évoluée avec toute une documentation en Français et incluant le fichier Python. Provenance : Courrier pour Windows 10 De : mgnthe54Envoyé le :jeudi 27 mai 2021 14:18À : Jyers/MarlinCc : tititopher68; AuthorObjet :Re: [Jyers/Marlin] Issue with new pack icons (include Gotcha icons and AxisBL, etc...) (#717) Bonjour titotopher68 (je suppose Christophe et Alscacien?) moi c’est Gilles et 54 pour la Meurthe et Moselle... Je suis débute dans le monde du 3D et je ne suis plus tout jeune...Super boulot que vous avez effectués, comme je le disais cela permet à la Ender 3V2 de réellement être beaucoup plus performante!J’utilise Cura pour découper mes fichiers STL.Je veux bien le lien du programme Python pour rajouter sur les fichiers gcode l’affichage de la ligne actuelle sur la totalité des lignes de la pièce...Je suppose que c’est un post traitement à appliquer après Cura (ou autre) sur les fichiers gcode. (on ne peut pas paramétrer directement Cura pour faire cette tache?)Effectivement la fonction M117 affiche un message à l’écran LCD d’après https://marlinfw.org/docs/gcode/M117.htmlL’affichage du layer se fera de façon fugitive sur l’écran ou indiquera en permanence le layer en cours? Ce n’est pas indiqué dans cette page web..J’ai rentré manuellement dans un fichier gcode un :M300 ; beepM117 Layer 0 ; texte et le Layer 0 s’affiche bien et reste présent à l’écran tant qu’on ne fait pas au moins un M117; sans rien pour efface le messageSi je peux un peu abuser dans les différents tuto Anglais d’où mes difficultés (mes 2 ans d’Anglais scolaires date de 45 ans...) on voit ajouter dans le Cura après le G28 qui remet la tête ne position de départ :soit G29 un nivellement du lit: linéaire, en 3 points,...soit M420 S1 active la mise à niveau! Enfin je sais pas trop ce que cela fait réellement, je suppose que l’on utilise la carte des niveaux que l’on a stockée depuis level –>Creat new Mesh (et leveling Active coché utile ou pas?) et qu’il y a une correction du zoffset en temps réel en fonction de la position de l’extruder?Quel est le mieux?Merci d’avancePS: une version Française du firmware est-elle un jour prévue et/ou une notice Française (à défaut en Anglais;;; avec google traduction et quelques essais pratiques) de ce que font réellement les fonctions de votre firmware pour les débutants comme moi?Merci encoreREPS: je suis équipé d’un Ender 3V2 carte mère 4.27 Dual Guard, BLTouch Creality et des derniers firmware 1.3.3 (25 points lent) et l”écran le custom