Jyers/Marlin #720: Live Adjust Going Wild, LCD Knob Response issues.

Chaseums0967, 2021-05-06:
Description

1) Live Adjustment causing axes to stutter back and forth when commanded to only flow in a single direction (AKA sending Z up so I can clean my bed.) Does not have the same issue when live adjust is off. Happens EVERYTIME I turn on live adjust.

2) When LCD knob is used to navigate screen, menu options often won't change until after the wheel has rotated a single 'click' twice. Seems only to happen after saving configuration and attempting live adjustments. Eventually causes screen to reset itself which usually clears any unsaved configurations.

Getting the same issues with both v1.3.1 and v1.3.2

Tthis is likely a "me" problem as I don't see anyone else talking about these issues. Curious as to whether or not there's some conflict between my LCD firmware and the printer firmware as my LCD is NOT running on standard Creality FW. Tho I did see on Reddit, Jyers commented that no LCD FW flash should be required for usage. Just scratching my head on what else it could be.

Steps to Reproduce

1. Select Live Adjustment, move any axis any direction (in a single direction).
**Expected behavior:** Axis smoothly moves that direction in real-time.
**Actual behavior:** Axis begins moving that direction, then will move a bit in the opposite, stuttering back and forth until it painstakingly reaches its destination.

2. Save config and/or select live adjustment. Check LCD knob for responsiveness. If not moving to other menu items from a single rotation 'click' then continue trying to adjust offsets and other menu items, saving config multiple times. Z home, autohome, level bed, basic machine setup in general. See if LCD screen ever errors out and resets itself.
**Expected behavior:** Config saves or live adjustment goes off without a hitch. LCD knob moves between menu items with a single rotational 'click' of the dial. No LCD reset.
**Actual behavior:** Knob becomes unresponsive to single 'clicks.' must move two 'clicks' before moves to next menu item. Issue persists until machine is rebooted. Goes back to normal after reboot.

#### Additional Information

Printer: Stock Ender 3v2 with stock LCD. 4.2.2 board. BLTouch v3.1 installed to mainboard.
Printer FW: .bin from Jyers repo, tried both v1.3.1 and v1.3.2 (current).
LCD FW: Honestly, can't remember at this point. Tried many different options in hopes it would allow me to live babystep my Z while printing, which wouldn't happen on standard Creality BLTouch firmware (the whole reason I embarked on this journey to begin with). Which is just one more good reason to believe it may be the LCD FW.

Can provide videos and additional info upon request.

More than willing to help out with this wonderful build of Marlin, just so new to all of it that I'm nearly positive it's user error due to my lack of knowledge.

tititopher68, 2021-05-06:
LCD knob has an issue, but it's a material issue. I also randomly have this. If I turn the knob very quickly in either direction, it seems to reset. I have already read this issue on other sites, not only for the Ender3V2 but also for other printers using this same screen ... even in stock firmware, I already had this ...
Creality must know the problem, because when I contacted support, they sent me a screen ... which worked fine for 1 month and again produced this phenomenon (and it was with another firmware). ..

Chaseums0967, 2021-05-06:
> LCD knob has an issue, but it's a material issue.

Interesting. I'm not too sure that on my ender it's a material issue tho. The knob consistently works totally fine until I try to take one of the above mentioned actions. And has never ever had an issue with slipping until now.

AngleV, 2021-05-06:
@Chaseums0967 

I was planning to report this issue as well as i have noticed this too also.

I would like to report also that this misbehavior happens usually when you rotate the encoder very fast.

Chaseums0967, 2021-05-06:
> I was planning to report this issue as well as i have noticed this too also.
>
> I would like to report also that this misbehavior happens usually when you rotate the encoder very fast as Christopher said.
> Honestly i do not know if it's faulty material or code related

The reason I believe it's code related is that I don't feel any physical mechanical slippages, and it only happens after certain actions are taken, and those actions consistently cause the issue.

Are you also experiencing the live adjustment problems?

Jyers, 2021-05-06:
The only way this would be a firmware bug is if somehow the encoder was getting put into rate adjust mode (increase encoder difference reports when spinning faster) I'll look into it but it seems unlikely. The note of "Eventually causes screen to reset itself which usually clears any unsaved configurations." really makes me think this is a hardware issue, as full reboot generally means a fatal error of some kind has occurred.

Chaseums0967, 2021-05-06:
@Jyers Copy that and thank you! Not trying to step on your toes here or make it seem like what you've done is faulty. It's almost certainly something on my end, I just don't know what that something is. 

Is there any possible chance it's that new LCD FW needs to be flashed? Sorry for any silly questions but this side of 3D printing is so new to me it's making my head spin. Just trying to spend the least amount of time troubleshooting/finding a fix action so I can just get back to printing, LOL. Haven't been able to for almost two weeks now.

Jyers, 2021-05-06:
@Chaseums0967 Sorry if it seemed like I was annoyed, this was like #8 of 12 notifications I can running though so I was just getting to the point 😅 You could try different screen firmware and see if it helps. At the moment I don't really have anything to go off besides checking the encoder rate, so feel free to update if you get any new information and I'll do whatever I can to help get this fixed.

Jyers, 2021-05-08:
Any updates on this?

firecrafty, 2021-05-08:
Just wanted to add on my experience- The LCD knob response issue only seems to happen on the home menu with the four blocks; it doesn't occur in sub menus. It also seems sporadic, most of the time the knob responds with 1 detent per menu item. I also noticed that even when the knob is responding properly the main menu responds to rotation slower than stock marlin firmware.

tititopher68, 2021-05-09:
@firecrafty you wrote "the main menu responds to rotation slower than stock marlin firmware." Yes of course ! because it's not the poor stock firmware....and the system is not a real multitaks and requires more execution time, but no choice if we want to benefit from features worthy of the name. Here, everything is perfectly structured in different readable and understandable module, not like the indigestible firmware of Creality. The button only jumps one step, essentially when the rotation is changed, without affecting a value change in a menu, etc. ;-)

firecrafty, 2021-05-09:
I was not making a criticism of the code- I was simply adding on my experience with this issue.

nilvanis, 2021-05-15:
I'm also experiencing the LCD knob response issue, mostly on the home screen.

STiBuki, 2021-05-21:
I am also experiencing similar encoder wheel issues on 3 separate printers. Two Ender 3v2's and a Voxelab Aquela (E3v2 clone) with extremely similar hardware, So much so that is can run the same firmware. It's defiantly related to acceleration to some extent. After some testing when scrolling at a steady state in any menu the cursor will accelerate or slow down on it's own and at times even reverse direction or not respond at all for one or two detent's on the knob. Whats the chances of all 3 having bad hardware? I have changed out the display on one of them with a spare display and ribbon cable with the same results. This all started happening with 1.3.0 rev and has gotten worse with each new revision since. Flash back to stock or any of the 1.2.x rev's and the issues are gone. I could see if it was just lagging in response but it's glitching much more than that. Also the enter function seems to be a bit overly sensitive. Whatever it is seems to be issues in general interpreting the encoder wheel inputs.

tititopher68, 2021-05-21:
I think you haven't analyzed the firmware code.  Understandable because
difficult and not necessarily within everyone's reach.
 There is no difference between the v1.2.x versions and now on the
management of the button, which is also the one present in the factory
firmware.

 The huge difference is the size of the firmware.  The hardware is not
multitasking and does not allow it to be done.  To simplify, the app runs
in a loop.  The app is much more substantial than the ultra basic one from
Creality.  Also, the more features there are, the more tasks there are to
accomplish, the longer the cycle times.  Not to mention the hardware limits
(the screen chosen is not the most efficient!) And so at times, decoding
the actions of the button causes the equivalent of a step to be lost ...
this is not the end  of the world given the current extent of the
possibilities to compare with the poverty of the factory version ...

firecrafty, 2021-05-21:
I don't want to be rude but your comments are coming off as very condescending and dismissive. The people commenting on this issue are simply sharing their experience with this issue.

I don't believe this issue is caused by lack of processing power. I'm not saying I know what the fix is, but simply saying fixing it is not possible is not conducive to the resolution of the issue.

Your comments are the type of comments that drive people away from open source development.

tititopher68-dev, 2021-05-21:
@firecrafty
 So you still haven't understood what I want to explain.  You seem to be taking this too personally!  You take this to be a kind of criticism of what is being said on the subject.  You are wrong !
 It is about explaining or trying to explain a behavior with this button precisely described and observed for a long time.  I don't have the science too.  And yes sometimes, there is not necessarily a solution.  I answered another problem on another subject by explaining why such a thing is not possible and that does not pose any problem.  It's just information!  And if you look a little more on this site, you will see that I am in the sharing, the proposal of fixes and rather listening to other users.  Except for you maybe, I could also flip your own hurtful remarks for once and I'm sorry.

Jyers, 2021-05-23:
Any updates on this? Is it still occuring with the latest release?

Chaseums0967, 2021-05-23:
@Jyers So sorry I haven't responded earlier!!! 

So, I haven't tried the latest release and more than likely will just stick with the v1.3.2. 

I never got live adjustment working, and the knob issues never went away, I've just been overlooking both issues (not using live adjustment and basically just doing as little navigation on screen as necessary to print). Haven't had time to continue to mess about with the issues unfortunately.

@firecrafty Have you tried the newest release by chance?

firecrafty, 2021-05-24:
I have not, I've been away for a few days but I should be able to test it tomorrow evening

Jyers, 2021-06-02:
Any improvement with more recent builds?

Jyers, 2021-06-09:
Closing due to inactivity, if we get any more info, or something new arises this can be reopened

closed 2021-06-09 by Jyers