Jyers/Marlin #743: Home individual axis from LCD crashes axis into endstop

firecrafty, 2021-05-09:
### Description

Homing individual axis from LCD does not work. Axis will crash into endstop and stall stepper.

### Steps to Reproduce

1. Compile and flash firmware using default-no probe config files
2. go to prepare -> homing -> home x (or any other axis)
3. Axis will start to home and crash into endstop 
4. 
**Expected behavior:** [What you expect to happen]

Axis homes properly

**Actual behavior:** [What actually happens]

Axis crashes into endstop

#### Additional Information

- "Home All" works properly
- "G28 X" (and all other axes) works properly
- "G28" works properly

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-05-09:
Hmmm, perhaps the home_axis function doesn't behave like I would expect. I'll switch it to a gcode parse for now.

Jyers, 2021-05-09:
Okay, should be fixed.

closed 2021-05-09 by Jyers

firecrafty, 2021-05-09:
Checked this morning- it's fixed. Thanks!