Jyers/Marlin #757: Not sure if issue or intended function.....

Gozania, 2021-05-09:
Currently on 1.3.2 I noticed that when you toggle the fan on in the temp settings, the fan that toggles on is the board cooling fan NOT this part cooling fan.   Is this the intended function?  If so where can I toggle on and off the part cooling fan?

asychev, 2021-05-09:
I do believe it is how Creality boards are _designed_ - the board fan and part fan are linked together and managed by the same PWM. This is why it is recommended to reconnect board fan to constant 24V supply.

Cal-LF, 2021-05-09:
Move the MB fan wires to the hot end fan terminals.
[image]

closed 2021-05-09 by Jyers

Gozania, 2021-05-09:
Thank you all for the input.  Thought something may have been wrong.  Alas its all by design.