Jyers/Marlin #772: Z endstop not recognised by Octoprint, causes crashing.

hippocratical, 2021-05-10:
### Description
Hardware/Software used:  
Ender 4 v2 using "E3V2-ManualMesh-3x3-v4.2.2.bin"
Latest version of Octoprint

Printing from SD card without Octoprint attached works great. 
Connecting USB from Octoprint to Printer, and sending print head to home crashes the print head into the bed.  

### Steps to Reproduce

With Octoprint connected:  

When not pressing the Z endstop I get this:

Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: y_min: open
Recv: z_min: open
Recv: filament: TRIGGERED
Recv: ok  

and this when manually pressing the Z endstop:

Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: y_min: open
Recv: z_min: open
Recv: filament: TRIGGERED
Recv: ok

So it's not registering the Z stop. Print head crashes.  

**Expected behavior:** [What you expect to happen]

For the Z endstop to be registered.

**Actual behavior:** [What actually happens]

Crashes into bed. Printer go Brr Brr.   

#### Additional Information
Chatting on Octoprint subreddit, people seem to think this is a Firmware issue rather than an Octoprint issue.

Now that I've isolated it to being a endstop issue, I found on Google people writing that you need the SD card in the printer.  
See here:  
https://github.com/MarlinFirmware/Marlin/issues/19223
https://github.com/MarlinFirmware/Marlin/pull/19226

Unfortunately I've tried both SD card IN, and SD card OUT (with both printer and Octoprint power cycled) and there was no difference. Z Endstop still not recognised.

Hardware/Software used:  
Ender 4 v2 using "E3V2-ManualMesh-3x3-v4.2.2.bin"
Latest version of Octoprint running on Windows 10

smooreace, 2021-05-10:
> MarlinFirmware#19223
> MarlinFirmware#19226 <https://github.com/MarlinFirmware/Marlin/pull/19226>

What version of Firmware are you using?

hippocratical, 2021-05-10:
> What version of Firmware are you using?  

E3V2-ManualMesh-3x3-v4.2.2.bin

I also tried the E3V2-UBL-NoProbe-5x5-v4.2.2.bin