Jyers/Marlin #784: Printer Error Serial status mismatch

araziforum, 2021-05-11:
Hi. In first thx for your kind project.
I'm using your firmware Ender 3 V2 (Manuel Bed Level 3x3 Mesh) with Octoprint v1.6 (also used 1.5). 
I'm using PSU control plug in with 15 sec post on delay settings. 
When I powered on my printer with octoprint I get M122 error on screen and Octoprint gets error like Printer Error Serial status mismatch. How can I fix it? or is it a bug?

Here is my log:

| Last lines in terminal:
| Changing monitoring state from "Offline" to "Detecting serial connection"
| Performing autodetection with 1 port/baudrate candidates: /dev/ttyUSB0@115200
| Trying port /dev/ttyUSB0, baudrate 115200
| Connecting to port /dev/ttyUSB0, baudrate 115200
| Handshake attempt #1 with timeout 2.0s
| Connected to: Serial<id=0xa30a73f0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
| Send: N0 M110 N0*125
| Recv: Error:Serial status mismatch
2021-05-11 14:16:51,078 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error"
2021-05-11 14:16:51,117 - octoprint.util.comm - INFO - Force-sending M112 to the printer
2021-05-11 14:16:51,181 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline after error"
2021-05-11 14:16:51,201 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-error' with value Error (Ender 3 v2)
2021-05-11 14:16:52,069 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=Error
2021-05-11 14:16:53,429 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-05-11 14:16:54,360 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-state' with value Offline (Ender 3 v2)

Tuncay-Ayhan, 2021-05-11:
I assume you mean M112(logs says so).

Try and remove your sd card and restart your printer. Problem should be gone.  It happens to me to when I leave my sd card in.

Also check https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

araziforum, 2021-05-11:
> I assume you mean M112(logs says so).
> 
> Try and remove your sd card and restart your printer. Problem should be gone. It happens to me to when I leave my sd card in.
> 
> Also check https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

Thx. It must be 112, my fault. I will try that.

Jyers, 2021-05-11:
Feel free to update if you continue to face this issue and I can reopen this