Jyers/Marlin #801: Thermal runaway error when manually stopping print job.. either Octoprint or LCD panel

cruzer619, 2021-05-12:
### Description
When I manually stop a print job.  I get a Thermal runaway error either from Octoprint or LCD panel

### Steps to Reproduce

1. [First Step] : print job
2. [Second Step] : cancel job on Octoprint or Stop on LCD panel
3. [and so on...] : Thermal runaway error shows up.

**Expected behavior:** [What you expect to happen]
Stop/cancel print job, print job stops and able to reprint.

**Actual behavior:** [What actually happens]
Stop/cancel print job, print job stops, but after 5 - 10 secs thermal runaway error beeps.  Turn off printer to do a reprint.

#### Additional Information
 using binary E3V2-BLTouch-5x5-HS-v4.2.7.bin version 1.3.2
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-05-12:
I'm still working on tuning the thermal runaway settings to provide the best balance of safety and flexibility. In the next release they will be adjusted to help alleviate such issues.

closed 2021-05-12 by Jyers

cruzer619, 2021-05-12:
ok thanks.. will wait for next release