Jyers/Marlin #829: Baudrate problem

umrotzer, 2021-05-14:
Is their a reason why u lowered the baudrate from 250000 to 152000 in your fw? What I see and read online, ender 3 v2 aso supports 250000 over serial what's more helpful for octoprint print over USB.

ComeHonourFace, 2021-05-14:
what is the baudrate?

umrotzer, 2021-05-14:
It's the serial speed to achieve between Ender and raspberry pi with USB.

On standard Marlin fw it's set to 250000 and in your fw it's set to 152000. Thats why I ask, if their was a reason for changing it.

I uploaded 2 pics, first is your configuration.h second is Marlin from your last release
[image]
[image]

t3chguy, 2021-05-14:
In the official firmware ( Ender-3 V2-Marlin-2.0.1-HW-4.2.7-mainboard-V1.1.2-compatible with BLTouch and filament detection ) it is set to 115200

[image]

The USB-Serial chip on the Creality boards is a CH340G - this one does not officially support 250000
https://arduino.stackexchange.com/a/43193

t3chguy, 2021-05-14:
Hell, even in Marlin's official example configs, it is 115200 for this board/machine.

https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20V2/Configuration.h#L130

closed 2021-05-14 by Jyers