Jyers/Marlin #838: Printer shuts off and starts up again.

Jeffaffa, 2021-05-15:
Hi,

On this firmware I've have an issue that the printer shutoff and starts up again.
The printer says that the power was interrupted and if I want to resume the print. This happened more often after I already completed a print and want to start a second one. 

I am back on firmware 1.3.1 and everything works fine. 
The board I have is a 4.2.7. I hope this feedback helps.

Jyers, 2021-05-15:
Most likely sounds like a hardware issue, but I'll leave this open until the next release so we can see if the issue persists.

Woodcat64, 2021-05-16:
This happened to me as well.

First I heard random clicking noise like it's skipping steps on extruder and while looking at it and trying to see where it's coming from. The printer just stopped and on the screen was a message something like "print was interrupted, do you want to continue?" We had no power loss, plus it's connected to UPS as well.

I said yes to continue and it just warmed up and started printing again.

This was on E3V2-UBL-BLTouch-10x10-v4.2.2.bin v1.3.1.  using micro SD to full SD card adapter. As a preventive measure I removed the adapter, but next print I got small layer shift on Z. This time I did the Creality "fix" where you ground the the chassis and Z axis stepper to the PSU. 

It has been good since, but still I have no idea what really happened. Of course I did not take a picture of the message. 
What is generating these messages? Is there a list somewhere I can go through?

Jyers, 2021-05-17:
A "looks like print was interrupted" message indicates the printer did reboot. Whether it was software or hardware related is the only question. But it's most likely an individual problem of some kind either way.

Friday13th87, 2021-05-18:
I have the exact same problem. Occured several times now, the first time while i was printing something really big at around 75% (wasted nearly 250g PLE O.o )
in my serial log is always only
2021-05-09 18:09:24,096 - Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3860
2021-05-09 18:09:24,100 - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2021-05-09 18:09:24,111 - Changing monitoring state from "Printing" to "Offline after error"
2021-05-09 18:09:24,121 - Connection closed, closing down monitor

i never had that problem before, only on FW 1.3.2

[config.tar.gz](https://github.com/Jyers/Marlin/files/6502009/config.tar.gz)

Jyers, 2021-05-18:
A new release should be coming out in the next day or so, so we'll see if it solves any of these issues

Jyers, 2021-05-23:
Give the newest release a try and see if you're still seeing the issue.

une18, 2021-05-23:
[image]
Started after upgrading for the latest ver 10x10 bl touch. Had not upgraded for few months. Printer works if i have it as offline printer. on error octoprint disconnects, abd this turns off the smart plug

Tuncay-Ayhan, 2021-05-23:
Remove your SD card if you want to use Octoprint.

une18, 2021-05-23:
I can do that, but had the card in the printer for last 6 months. Upgraded yesterday and then the problems started. Some breaking changes in the latest builds?

Tuncay-Ayhan, 2021-05-23:
It’s a know issue with Marlin. Check the Wiki pages > Octoprint

Jyers, 2021-05-23:
https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

This details how to disable M112 messages to avoid such issues

une18, 2021-05-25:
@Jyers  That fixed it! Thank you!

Jyers, 2021-05-26:
Great! I'll close this for now and if you see the issue again, just update this and I can reopen it.

closed 2021-05-26 by Jyers