Jyers/Marlin #842: Pressing Cancel after a Filament Change does not close the dialog

nullstalgia, 2021-05-16:
Also, that pre-heat menu is *nice*. But may I recommend having Continue at the top under Back? Since it seems kinda low for most people who are going to use the top slot PLA, also its hidden for first time users, but if they scroll down there it is. Just a random suggestion. :P

Jyers, 2021-05-16:
Which popup are you referring to? The purge more popup or like the filament change confirmation?

Also you're totally right, continue would be better at the top :)

nullstalgia, 2021-05-16:
The purge more, regardless if I do purge more or not

nullstalgia, 2021-05-16:
But maybe you could put a second **Continue** on the bottom for people who use Custom? Unless when you set it, it also resets the cursor to the top. (I haven't tried it, just expecting it to work like the Move XYZE menu)

nullstalgia, 2021-05-17:
Also, I think the "Please Wait" pop up after clicking "Continue" on the "Heater timed out" on the filament change is missing..

Jyers, 2021-05-17:
I'll take a look at that

tititopher68, 2021-05-18:
fixed with : 
case PurgeMore:
          if (selection==0) {
            pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
            Popup_Handler(FilChange);
          } else {
            pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT;
            if (printing)   Popup_Handler(Resuming);
            **else   Redraw_Menu(true, true, (active_menu==PreheatHotend));**
          }
          break;

Jyers, 2021-05-18:
Thanks @tititopher68, fix has been pushed

closed 2021-05-18 by Jyers

nullstalgia, 2021-05-18:
@Jyers 

"Also, I think the "Please Wait" pop up after clicking "Continue" on the "Heater timed out" on the filament change is missing.." this is still occuring. Should I open a new issue for it?

Jyers, 2021-05-18:
Oh yeah, don't worry I'll do that now.

Jyers, 2021-05-18:
Done 🙂

tititopher68-dev, 2021-05-18:
> @Jyers
> 
> "Also, I think the "Please Wait" pop up after clicking "Continue" on the "Heater timed out" on the filament change is missing.." this is still occuring. Should I open a new issue for it?

No... I don't understand...I tested and the popup " Please Wait..." isn't missing after clicking "Continue" at the end of "Heater...."

nullstalgia, 2021-05-23:
@Jyers Sorry, but the issue is still present.

https://streamable.com/awgfbd

The only files I have modified are Config and Config_adv

Jyers, 2021-05-23:
Oh damn, I totally read your message wrong. I'll fix that tonight or early tomorrow. It's an easy fix

Jyers, 2021-05-23:
@nullstalgia Let me know if the latest commit fixes the issue

nullstalgia, 2021-05-23:
Bingo 👌 

Thanks again