Jyers/Marlin #850: Update of Display Firmware Readme.md

tubalainen, 2021-05-17:
Would it be possible to please update the readme.md file explaining the different folders and which one to pick? :)
Thanks.

[image]

joeclifford, 2021-05-17:
There is s discussion about this.

https://github.com/Jyers/Marlin/discussions/794#discussioncomment-725172

Should be easy to add to the wiki.

tubalainen, 2021-05-17:
> There is s discussion about this.
> 
> [#794 (comment)](https://github.com/Jyers/Marlin/discussions/794#discussioncomment-725172)
> 
> Should be easy to add to the wiki.

Yes. I am all for making 3D printing as easy as possible. There´s enough of hurdles out there already. Thats the reason.
I will make a PR for the readme.md file and also one for the Wiki.

How ever, I cannot still understand of there is a difference between Gotcha and Custom.

smooreace, 2021-05-17:
Dire need for a :facepalm: emote!

tubalainen, 2021-05-17:
> Dire need for a 🤦 emote!

Agreed - I hate when there is a lack of "emotes". 

I do not see how helping other can be such a bummer. Its people like you that so efficient puts out all the sparks of open source and collaboration project. Please elaborate, lift, inspire, collaborate instead of just being snarky and irrelevant.

tome9111991, 2021-05-17:
Go in Icon Packs folder.
You can see the .jpg of icons

Jyers, 2021-05-17:
It's already on the wiki, I updated the display firmware section a few days ago :) https://github.com/Jyers/Marlin/wiki/Configuring-&-Flashing

closed 2021-05-17 by Jyers