Jyers/Marlin #857: New Icons

Lighty1989, 2021-05-17:
Hello Jyers, i created new icons

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jyers, 2021-05-18:
I very much like the new icons, but unfortunately I don't think I'll be able to use manual leveling ones, as there is only so much space for icons on the display and the ones used currently are decent. I'll definitely keep these in mind for any new icon packs though, as I love the x, y, and z. What are the light looking ones for btw?

closed 2021-05-18 by Jyers

Lighty1989, 2021-05-18:
Laser symbol on / off, I'm still looking for a solution to turn the laser on very weakly. anycubic somehow got this solved. so that you can position yourself better. x y z icons are for the very bottom and the others for manual leveling