Jyers/Marlin #860: v1.3.2 - no control screen during printing

czaras, 2021-05-18:
I've installed to my Ender 3v2 this firmware: E3V2-ManualMesh-3x3-v4.2.2.bin

When I start printing the printer screen shows a regular print screen with Tune button etc. but after a few minutes the display goes back to home screen and there's no option to display a control screen again althought the printing is still in progress, you're not able to pause or tune the current ongoing job.

I did reinstall this firmware twice but the story is still the same.

tititopher68, 2021-05-18:
@czaras 
Have you tried compiling your own firmware with the latest build?

SikVac, 2021-05-19:
I have the same problem. When printing starts, the screen switches to the home screen.
I solve this by switching to the Tune option immediately after starting printing. During printing, you can then select the Back option again, which puts me on the print screen.

czaras, 2021-05-19:
ok, good to know the temporary solution ;) but it’s an obvious bug, right? ;)

         
Powered by Mailbutler <https://www.mailbutler.io/?utm_source=watermark&utm_medium=email&utm_campaign=watermark-essential-email>, the email extension that does it all

tititopher68-dev, 2021-05-19:
@SikVac @czaras Sorry but, no issue with the last build on line. You have to compile your own firmware...

Cal-LF, 2021-05-19:
The next release will fix this issue. I had the same thing happen and compiling the 3X3 man mesh template fixes it.

Qubi-B, 2021-05-21:
does it do the same with 5x5 manualmesh or without manual mesh at all?

SikVac, 2021-05-22:
To Qubi-B
With version "E3V2-UBL-BLTouch-15x15-HS-v4.2.2.bin"

Jyers, 2021-05-22:
As a temporary fix, I believe this is caused by having M501 in your start gcode. Removing this should fix the issue. As @Cal-LF said though, this has been fixed and will be in the next release

closed 2021-05-22 by Jyers