Jyers/Marlin #869: Font

Qubi-B, 2021-05-20:
Can I change the font used in fw? I would like to make it more minimal. If it's possible, then how to do this? And if it's not, will it be possible in the future?

Sweyn-Forkbeard, 2021-05-20:
I agree, a Sans Serif font would be cleaner and clearer

adam3654, 2021-05-20:
I'm pretty sure the font is baked into the screen firmware file and isn't editable. I've poked around a bit and anything font related in the source code seems to be for the lcd screens. If anyone knows otherwise I'd love to hear about it, I can't stand the font either.

adam3654, 2021-05-20:
> @adam3654 Exactly ! Not editable...

@tititopher68 Has there been any feedback from creality as to why the screen firmware is closed source? I have searched and found nothing.

Sweyn-Forkbeard, 2021-05-20:
I have an Ender 3 v2, I should have noted that, my understanding is that the screen can be flashed from a micro SD slot on the back of the screen, I may well be chewing on the wrong end of the stick, and I'm tired so my thought process has sort of shut down.

tititopher68, 2021-05-20:
This display isn't a specific Creality Display but a DWIN display : https://beijingdwin.en.ec21.com/

Qubi-B, 2021-05-21:
so we are unable to do anything? maybe something like custom char set or make a sans serif font into normal display icons and then make it change letters to related icons? idk if its possible but thats my thought

tititopher68-dev, 2021-05-21:
The screen fonts are integrated directly into the .bin file in the DWIN_SET folder.  It is impossible to modify it, because there is no source or documentation for it, including on the Chinese site of DWIN, the documentation is moreover strictly reserved ... I only received from technical support one  latest version of this .bin file which changed from v20 to v23 but without further information.

smooreace, 2021-05-21:
Ide imagine that someone capable could dig through the bin file and find
the font assignment. But I can imagine how tedious that might be!

tititopher68-dev, 2021-05-21:
The Chinese from DWIN just sent me this:
T5UIC1 is based on Diwen Technology T5 CPU, a simplified version designed for applications that do not require a touch screen, simple UI functions, and demanding cost requirements.
 Serial command screen.
 Its main features include:
 (1) 65K color TFT display.
 (2) Basic drawing commands, Chinese and ASCII text display, support JPEG icon, JPEG picture, bar code, two-dimensional code display.
 (3) 384Kbytes font space.
 It stores 6*12-32*64 dot matrix ASCII and 12*12-64*64 dot matrix GB2312 Chinese character library (Chinese characters are scaled based on 16*16 dot matrix).
 (4) 512Kbytes image and icon storage space is divided into 16 storage spaces according to 32KB.
 It can store up to 16 JPEG full-screen pictures.
 Or store 0-16 JPEG icon library files (a single icon library file can exceed 32KB and occupy multiple memory spaces).
 (5) 32KBytes SRAM data memory that can be read and written by serial port, the data is lost after power-on, and all initialized to 0x00 after power-on.
  Mainly used in online pictures, icon library data update, or real-time JPEG icon, picture display.
 (6) 16Kbytes Flash data memory that can be read and written by serial port, the data will not be lost when power off, and the write life is 100,000 times.
  Mainly used for data storage such as user configuration parameters.
 (7) SD/SDHC interface configuration parameters and update fonts and pictures.
 (8) An additional full-duplex serial port is extended.
 (9) The CPU can be configured to run at 250MHz or 400MHz.

smooreace, 2021-05-21:
Clearly the font can be modified then!

tititopher68-dev, 2021-05-21:
I just got some more info.  There is 1 bit at an address in the .bin for managing character spacing.
 I received a utility to extract the FONT files from the .bin and re-encode it.  I'll see that tonight.  We can a priori update the FONT via a 0T5UIC1.HZK file in the DWIN_SET folder ...
 But for the moment no information for the shape and structure of FONT ... To be continued ..

smooreace, 2021-05-21:
If you want to share what youve received I can poke at it

Sweyn-Forkbeard, 2021-05-21:
I'm disappointed that I can't help more, this was just my thing in the 80's. Sadly a brain trauma has robbed me of the ability to sit and process lines of code. You have no idea how many times I've had to use spell correct. It's frustrating. Good to see group minds working together and unpicking the framework. I wish you luck.

gothcha, 2021-05-22:
Some info here about the font tool

https://www.gitmemory.com/issue/MarlinFirmware/Marlin/18825/666142596

Jyers, 2021-05-22:
I might actually be able to create new .HZK files with different fonts. While I was working with one of crealitiy's new touch screens, I notices that in the DGUS tool there is a way you can modify the font. If that same file can be applied to this display, I might be able to put any font you want on it :) 

I'll take a crack at building a custom font and send over the files for anyone interested in testing it.

smooreace, 2021-05-22:
The link that @gothcha posted has the tool to edit change the fount and build the .hzk file

https://www.gitmemory.com/issue/MarlinFirmware/Marlin/18825/666142596

Jyers, 2021-05-22:
@smooreace Ahh yeah, I didn't see that, That's exactly the same one found in the DGUS tool

Sweyn-Forkbeard, 2021-05-22:
I would certainly like to test drive any new font. As I said I'm disappointed I have lost the ability to code, it still fascinates me, My brain though just blows a fuse. I've been doing it since the mid 70's, perhaps it's time for a rest....

Qubi-B, 2021-05-22:
> 
> 
> I would certainly like to test drive any new font. As I said I'm disappointed I have lost the ability to code, it still fascinates me, My brain though just blows a fuse. I've been doing it since the mid 70's, perhaps it's time for a rest....

I'm really sorry abut that.. What happened?

smooreace, 2021-05-22:
Fortunately, no coding seems to be required

Sweyn-Forkbeard, 2021-05-22:
> > I would certainly like to test drive any new font. As I said I'm disappointed I have lost the ability to code, it still fascinates me, My brain though just blows a fuse. I've been doing it since the mid 70's, perhaps it's time for a rest....
> 
> I'm really sorry abut that.. What happened?

Chronic and degenerative frontal brain trauma caused in a cycle crash whilst in the chasing pack, yes I had a helmet, they shatter when you bounce off the ground at 80kmh. I now ride around in different wheels (spinal injury) or stagger about and fall over a lot, much to the amusement of my best buddy, more because I'm laughing so much myself. It hasn't stopped me doing much, language still works to a degree, I am thankful of spell check, and I have set myself up for two world records bulling heavy vehicles whilst in a wheelchair. I compete (or did before Covid) in road racing, wheelchair style. There is still an incredibly determined and independent person who just happens to have wheels for legs. Tomorrow I'm going to help my buddy remove the rear axle of his Volvo 240 for restoration. I have my crawler board, once I'm down that's it. Live life to the full, tomorrow isn't guaranteed.

smooreace, 2021-05-22:
>
> Live life to the full, tomorrow isn't guaranteed.
>

That right there is really the key to life!

Lighty1989, 2021-05-23:
https://github.com/odwdinc/DWIN_T5UIC1_LCD

Qubi-B, 2021-05-24:
> 
> 
> I might actually be able to create new .HZK files with different fonts. While I was working with one of creality's new touch screens, I noticed that in the DGUS tool there is a way you can modify the font. If that same file can be applied to this display, I might be able to put any font you want on it :)
> 
> I'll take a crack at building a custom font and send over the files for anyone interested in testing it.

Please let us know when u have the files!

Jyers, 2021-06-02:
Let's move this to #953

closed 2021-06-02 by Jyers