Jyers/Marlin #870: Mise à jour V1.4.3.a;

Venom1981, 2021-05-20:
Bonjour, pouvez me dire ou je pourrais trouver la mise à jour V1.4.3.a:?
Merci pour votre aide.
Cordialement.

tititopher68, 2021-05-20:
Bonjour, ce n'est une version du firmware originale de Jyers, mais la mienne sur ma page Github : https://github.com/tititopher68/Marlin

Venom1981, 2021-05-20:
Pourriez-vous me donner le lien direct car depuis ce matin je cherche mais je n'arrive pas à trouver,🤔je suis vraiment désolé 🙏

tititopher68, 2021-05-20:
La version stable est celle ici : https://github.com/tititopher68/Marlin. En version 1.5.0.f maintenant. A vous de la compiler en suivant le PDF dans le dossier Documentation...

Venom1981, 2021-05-21:
D'accord merci, par contre dernière question, dans la documentation il y a écrit qu'il y a une modification de la carte maire à faire, c'est quoi comme modification ?
Désolé mais je débute 😔

Jyers, 2021-05-22:
Please keep fork related issues off of the main repo. Thank you!

closed 2021-05-22 by Jyers