Jyers/Marlin #901: 1.3.3 Preheat options do not reset unselected temperatures

gothcha, 2021-05-25:
Jacob something I noticed when working with the new 1.3.3 (likely also for earlier versions)
When a Preheat option (Both, Hotend, Bed) is selected and the Preheat PLA clicked the unselected heaters are not reset to zero.
e.g if Preheat both was initially set and the bed and hotend heaters are active, changing the option to Bed and reselecting PLA leaves the hotend at the previous value. Currently the only option is to use cooldown to reset the heaters/fan, then go back into Preheat.

This can be fixed by resetting the heaters/fan before the new setting is actioned.

            case PREHEAT_1:
              if (draw) {
                Draw_Menu_Item(row, ICON_Temperature, PREHEAT_1_LABEL);
              }
              else {
                _thermalManager.disable_all_heaters();
                thermalManager.zero_fan_speeds();_
                if (preheatmode == 0 || preheatmode == 1) {

Now all the temps/fan speed are set correctly obeying the current mode.

Jyers, 2021-05-25:
Hmmm, I actually never considered that an issue, but you're definitely right, that functionality would make more sense. I'll fix it when I get home

Jyers, 2021-06-02:
Implemented, thanks again!

closed 2021-06-02 by Jyers

Cal-LF, 2021-06-03:
This works great!