Jyers/Marlin #902: Manual Bed Leveling Disable M420

eblieb, 2021-05-25:
Need to ensure M420 is set to M420 S0 before manual bed leveling or else it will mess up the bed leveling every time. I verified by running the bed leveling then manually setting M420 S0 and telling it to go to the same point and the Z height was different after disabling the leveling.

Jyers, 2021-05-25:
`#if HAS_LEVELING`
`      level_state = planner.leveling_active;`
`      set_bed_leveling_enabled(false);`
`#endif `
`Draw_Menu(ManualLevel);`

I can assure you this is already the case. If the z height changes it must be from a different factor. What leveling system are you using?

eblieb, 2021-05-25:
Where in the code is that? Using a bltouch with abl

Jyers, 2021-05-25:
https://github.com/Jyers/Marlin/blob/c4c58e3317045e3331c85abcb30ea4d425986f07/Marlin/src/lcd/dwin/creality_dwin.cpp#L1077

eblieb, 2021-05-25:
Strange.. Must have been a glitch then

 

https://github.com/Jyers/Marlin/blob/c4c58e3317045e3331c85abcb30ea4d425986f07/Marlin/src/lcd/dwin/creality_dwin.cpp#L1077

closed 2021-05-25 by Jyers