Jyers/Marlin #903: Printing in the Air - BLTouch - Mesh

Runortyx, 2021-05-25:
Hey there,

i have flashed the E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.3.bin to my Ender 3 v2
But other Firmwares before, ich got the Problem that the Bed is Manual Leveld, after that i checked the offset, and after that i mesh leveld the bed.
If iam printing, he is printing in the air, the Nozzle is way to high, nearly 2cm, but if i tune the print and level again or somtheing its way to down and nozzle is crashing in the bed.

dont know what the problem is, i pulled the cable from the endstop Z because i have bltouch.

Jyers, 2021-05-25:
It could be an issue with probe reliability. Try the z probe repeatability test in Control>Advanced>Probe

domeOo, 2021-05-27:
I am experiencing exactly the same thing. I will test and report with an older fimrware verison

georgesir, 2021-05-31:
Hi so I am experiencing the same problem, it seems that the prop is off by as much as 0.4 in some cases. When I go to probe tuning and go to actual z offset value its way out. Have you guys found a solution for this? 

Running Ender 3v2, board 4.2.2 3x3 bl touch hs/ns and 5x5 hs/ns

I have the cable still connected to endstop z switch as well as bl touch.

AllanMadsen, 2021-06-02:
> I have the cable still connected to endstop z switch as well as bl touch.

My Bl Touch install install guide clearly states that Z endstop must be disconnected. Othervise Bl Touch will not work properly.

plucker, 2021-06-02:
I doubt that leaving the cable connected to the motherboard can make any difference, but the microswitch should be removed

Jyers, 2021-06-02:
Any updates on this?

domeOo, 2021-06-02:
i installed the previous version and upgraded again after calibrating and creating the mash with the bl-touch. No it works. I don't know why, but it worked for me. Maybe it is worth a try @Jyers @Runortyx

everdom, 2021-06-09:
I am also experiencing the same problem, I adjusted the height of the nozzle through the Z Offset menu so that it just touched the bed, but when printing, I found that the nozzle was about 0.35mm higher than the bed, so I had to manually adjust the Z Offset down 0.35mm, but I didn't think that was the correct way to solve the problem

closed 2021-06-09 by Jyers