Jyers/Marlin #909: Issue create mesh with changed Z step valu

tome9111991, 2021-05-26:
Hello
I dont know is a issue with firmware or is a general problem.
I use bltouch with 5x5

I have Z step value 400 (default)
Have no problems to creata mesh looks like:
[image]

In motion settings i change Steps > Z from 400 to 408
I start auto level and after 13 from 25 ponts i got the message auto level finish.
My mesh looks like
[image]

Have nothing change else.

smooreace, 2021-05-26:
I would expect some change, but thats pretty big!

Jyers, 2021-05-28:
Have you tried multiple times? A probe failure during leveling could probably cause such issues.

tome9111991, 2021-05-28:
Yes i have tried multiple times.
I had never a probe failure during leveling before.

can someone possibly reproduce this?

Jyers, 2021-05-31:
Any updates on this? Also, why do you need to change your Z steps? it should be cut and dry if your using a lead screw setup.

smooreace, 2021-05-31:
Changing steppers or changing to a more narrow pitch lead screw would
require a change in Z steps. Ive seen several people wanting to change
pitch to compensate for insanely overweight hot-end cooling solutions with
DD

tome9111991, 2021-05-31:
> Any updates on this? Also, why do you need to change your Z steps? it should be cut and dry if your using a lead screw setup.

It was only a test.
I change back to default  400 no issues

Jyers, 2021-05-31:
@tome9111991 My best guess is that since the printer wasn't moving the distance it thought it was, it caused some issue with the probe. If this ever becomes an issue with a physically different z setup, we can reopen this

closed 2021-05-31 by Jyers