Jyers/Marlin #910: M900 doesn't work with 1.3.3 code. Says "unknown command"

mecra, 2021-05-26:
### Description

### Steps to Reproduce

1. M900 code says unknown command.
2. https://reprap.org/forum/read.php?415,865194

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mecra, 2021-05-26:
https://reprap.org/forum/read.php?415,865194 

Someone posted a solution here to a missing linear advance config.

JoelMajor, 2021-05-26:
Not sure, but if your printer as the 4.2.2 board you're not able to use Linear Advance.

https://www.reddit.com/r/ender3v2/comments/hsp7jn/ender_3_v2_with_linear_advance/

mecra, 2021-05-26:
If the calibration test actually prints, doesn't that mean that it works?

Matthew Larson

https://www.reddit.com/r/ender3v2/comments/hsp7jn/ender_3_v2_with_linear_advance/

Giadej, 2021-05-27:
> 
> 
> Not sure, but if your printer as the 4.2.2 board you're not able to use Linear Advance.
> 
> https://www.reddit.com/r/ender3v2/comments/hsp7jn/ender_3_v2_with_linear_advance/

https://github.com/Jyers/Marlin/discussions/707

Giadej, 2021-05-27:
> 
> 
> Not sure, but if your printer as the 4.2.2 board you're not able to use Linear Advance.

https://github.com/Jyers/Marlin/discussions/707

Jyers, 2021-05-28:
Are you compiling your own firmware? "Linear Advance Support" in the release notes doesn't mean all builds have linear advance, it just means there is support if you want to enable it in your own build

mecra, 2021-05-28:
There was another guy who made firmware that got linear to work on 4.2.2. Can't remember who it was. But no I didn't recompile. I thought it was enables by default in the new firmware.

Matthew Larson

Schnabulation, 2021-05-28:
The thing is: Some 4.2.2 boards come with stepper drivers that support linear advance out of the box. But it's not clear which boards these are.

If you enable linear advance in the firmware (by compiling yourself) and then can print the linear advance test print then you are in luck!

Giadej, 2021-05-28:
> 
> 
> The thing is: Some 4.2.2 boards come with stepper drivers that support linear advance out of the box. But it's not clear which boards these are.
> 
> If you enable linear advance in the firmware (by compiling yourself) and then can print the linear advance test print then you are in luck!

The 4.2.2 cards to allow linear advance are those that mount the TMC2209 drivers. It is possible to identify the driver mounted simply by reading the letter written (with marker) on the card reader slot.

[image]

https://github.com/Jyers/Marlin/discussions/707#discussioncomment-701286

[image]

Schnabulation, 2021-05-28:
Ah yes, you are right - there was something about the letter on the SD slot. However: my board did not have any letter on it and linear advance did not work.

closed 2021-05-28 by Jyers