Jyers/Marlin #916: Good evening. After updating from 1.3.2, the first layer started to appear under-extruded, and in the second layer it had a bizarre behavior. It would print well in the furthest Y direction, and under-extruded in the closest Y direction. I had my bed leved, and tryed different files. I've leveded the bed every time. Went back to 1.3.2, no problems!

rodolfopaulo, 2021-05-27:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-05-28:
After updating to 1.3.2 did you reset eeprom? After flashing go to the control menu and select reset to defaults and then store settings.

Jyers, 2021-05-31:
Closing due to inactivity. Feel free to update this and it will be reopened.

closed 2021-05-31 by Jyers