Jyers/Marlin #918: E3V2-BLTouch-5x5-HS-v4.2.2-v1.3.3 M190 S{material_bed_temperature_layer_0} requires reboot after print or consequences

branlan, 2021-05-27:
### Description

When using this gcode:
`; Ender 3 Custom Start G-code
M92 E98 ; EStep offset
M190 S60 ; preheat bed to 60
G28 ; home
G29 ; BLTouch
M190 S{material_bed_temperature_layer_0} ; heat to setting 
M109 S{material_print_temperature_layer_0} ; heat to setting
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish`

One print will finish normally, but if you start a second print it will do the G29 command, preheat, but then will just start extruding filament on the spot. [see video here ](https://photos.app.goo.gl/tfhNk7cXegr8ztBu6)

Tested without the:
M190 S{material_bed_temperature_layer_0} ; heat to setting 
M109 S{material_print_temperature_layer_0} ; heat to setting
lines in my gcode, and the issue does not occur. 

### Steps to Reproduce

1. Update firmware to 1.3.3
2. Add M190 S{material_print_temperature_layer_0} line to start gcode, slice and move to printer
2. Print and let finish
3. Start another print without cycling power
4. When the BLTouch finishes leveling and the nozzle has preheated, watch the filament spew out of the nozzle.

**Expected behavior:** 
First print should finish, and then subsequent prints should work normally.

**Actual behavior:** 
First print finishes normally, all subsequent prints fail and spew out filament from the last place the BLTouch leveling process left off.

Jyers, 2021-05-28:
Wow, very strange issue. Is this printing from OctoPrint or SD card? Also did you clear eeprom after flashing the firmware?

branlan, 2021-05-28:
Printing from an SD card, and yes I did clear the eeprom after flashing the new firmware! 

It is very strange, I'm wondering if somehow the 

> M190 S{material_bed_temperature_layer_0} ; heat to setting
> M109 S{material_print_temperature_layer_0} ; heat to setting

lines are being persistent through prints instead of clearing after the first print finishes, but I'm not familiar enough with gcode to know how to test that.

Jyers, 2021-06-01:
I still can't imagine what would be causing this. Perhaps you could try moving the preheat commands to before your G28 and see if that helps.

branlan, 2021-06-02:
Moving the preheat commands before G28 / removing them entirely does solve the problem of not printing / extruding on one spot on second print if not restarting the machine, however that defeats the purpose of why I put those in the startup gcode; 

The purpose of this was to preheat the bed and have the bltouch autolevel prior to the nozzle heating up to prevent the filament from coming out of the nozzle while the bltouch was doing its leveling. 
I guess it doesn't really matter that much and I can just remove it, however it's odd it happens to me on this firmware on not on the stock ender3v2 firmware.

Jyers, 2021-06-02:
Have you tried just moving just the bed preheat in front of G28? That would save you the issue of heating the hotend and possibly isolate the issue to just the bed preheat. It's worth a shot at least.

domeOo, 2021-06-02:
I installed the previous version and upgraded again after calibrating and creating the mash with the bl-touch. No it works. I don't know why, but it worked for me. Maybe it is worth a try

branlan, 2021-06-05:
I ended up adding a bit more retraction in my end gcode to fix the drizzling I was having, and removed those preheat commands.

closed 2021-06-05 by branlan

Jyers, 2021-06-05:
So strange, sorry we never figured it out fully.