Jyers/Marlin #922: Feature Request -- Hight Temperature Thermistor Upgrade

HermanGaviriaMarin, 2021-05-28:
Hello Grand Master Jyers (or whoever is reading),
I am interested in printing some exotic filaments that require temperatures that are above the stock max temperature (260 C). I understand that this is for safety reasons concerning the Bowden PTFE tube, to which I got an all metal hot end for. I have been told that in order to have print at higher temperatures I would have to compile my own version of your firmware, the truth is that I have tried doing so before and things didn't go too well, for that instance I was not messing with the max temperature settings. I believe that your expertise suits you better for the task, specially if its something as delicate as max temperature settings. If at all possible a version of your firmware that allows us to print at higher temperatures either with the use of a high temperature thermistor (300 C temps) or with the use a thermocouple, would help a lot. Thank you for your time.

PS. With regards to which type of thermistor or thermocouple to code for, I would say just pick the one that is easiest for you or the community to use, just as long as we are able to hit temperatures like 300 C or just a little bit past that. I understand that stock hardware wont work for this.

smooreace, 2021-05-28:
As I just mentioned... YOU compile your firmware for what hardware YOU want
it to work with.All strains of Marlin are already capable of taking
whatever thermister you chose to compile with.

HermanGaviriaMarin, 2021-05-28:
NOOO LMAO..... I'm scared to fuck things up. But if I must I will.

smooreace, 2021-05-28:
It's your responsibility to compile firmware the way you want it. That's
what makes open source firmware.. or anything for that matter, so great. It
gives US the power to change direction is needed. All Jyers is doing is
giving us more control of out screens. Obviously not even slightly related
to fundamental changes in hardware! Fortunately to you, and many of the
rest of us here, Marlin gives us the ability to manipulate the tools as we
need them.

ubeyou, 2021-05-30:
https://patreon.com/Jyers

you can just pay Jyers $5 per month to access the auto compile service. It's cheaper than a filament.

HermanGaviriaMarin, 2021-05-30:
Would this make it easier for me to create those changes ? Or is this just a way to not have to use visual studio code ? Also while in visual studio code there was no pre-set for the 5x5 mesh with BL touch, only 3x3, is there a simple way to change this ?

smooreace, 2021-05-30:
Search for "probe" in configuration.h and look around. X probe frequency is
defined as a # and Y is currently defined as equal to X

Jyers, 2021-05-31:
@HermanGaviriaMarin The compile service technically only helps with not needing visual studio, but you also get access to the discord, which makes it much easier for me to help you withy configuration details if you need :)

As far as adding a precompiled for this, I still want to keep the amount of precompiled version to a reasonable level and adding a new maxtemp would double them. So for this reason adding a new skew for higher temps isn't really viable.

HermanGaviriaMarin, 2021-05-31:
Grand Master Jyers, I will get this service you speak of. I seek your guidance, I believe I can finally change what I need to change but there are other small details that require your expertise or that of other discord members. Anyways thanks for the direct response, I know you must be a busy man

Jyers, 2021-05-31:
Glad to hear it brother Herman. Godspeed