Jyers/Marlin #928: Strange issue with bltouch not wanting to probe after reflashing back to jyers from klipper on Ender 3 v2

DrConflict, 2021-05-28:
Description:

When trying to home all axis after reflashing back to jyers from klipper the z axis does not go down. probe goes to middle of bed, extends multiple times and then just stays there. Screen reads: STOPPED.

### Steps to Reproduce

1. Flash jyers 1.3.3 to ender 3 v2 with a fully working fluidd/klipper setup.
2. Try to home all axis.
3. x and y axis home fine, then bltouch goes to middle of bed and extends multiple times. Then just stays there. The z axis never goes down.

**Expected behavior:** [What you expect to happen]

It homes all axis fine.

**Actual behavior:** [What actually happens]

#### Additional Information: Running all latest versions of jyers.

(I can not provide config files since i flashed the stock 5x5 bltouch 1.3.3 jyers for the 4.2.2 board..

smooreace, 2021-05-28:
You cant run BLT and have the Z stop at the same time without custom
compiled options

DrConflict, 2021-05-29:
@smooreace Not sure what you mean with that. I don't have the z endstop attached to the printer anymore since I have been using the bltouch for a long time.

smooreace, 2021-05-29:
Thats what typically causes what you are describing

vhristenco, 2021-05-29:
@smooreace I have both the bltouch and the z-stop plugged in and it runs without any issues.

DrConflict, 2021-05-29:
@smooreace Yeah but why would I have a Z stop endswitch on my machine when my bltouch acts as the z endstop. I took it off once i installed my Bltouch.

smooreace, 2021-05-29:
@DrConflict ... a lot of people seem to make the mistake. And generally, its when they are new and have just installed the BLT. And, some people have their BLT plugged into the Z stop port... no clue. Was just an idea

@vhristenco ... the only way that would be working is if the z-stop is so far below the point where the BLT triggers that its never tripped... which would cause the problem Conflict is having

vhristenco, 2021-05-31:
@DrConflict actually the z-stop is mechanically engaged every time the printhead goes to Z=0, but it's ignored since uncommenting #define USE_PROBE_FOR_Z_HOMING in the configuration.h
Haven't had any issues with this so far, and I'm using the BLTouch for about an year...

closed 2021-05-31 by Jyers