Jyers/Marlin #930: Progressbar jumps from 0 to 100 and Heating stops

MStinso, 2021-05-29:
Hello everybody,
when i start printing, the printer heats up normally and then starts printing. After about 1 minute of printing, the progress bar jumps to 100% and the confirm window appears (actual printing time 4 hours). The printer stops heating and continues to print. after clicking on confirm it does not stop either.
I have already tested several slicers and different firmware versions.

I have tested all firmware versions (1.1.0-1.3.3) and have the same problem everywhere.

When I use the Octoprint printer, the same thing happens and the connection to the printer is disconnected.

The original Ender Firmware works without a problem.

My Printer: 
Ender 3 V2 4.2.2
Creality BL-Touch

i would be happy about a solution to my problem.

best regards

Jyers, 2021-05-31:
Based on the problem and testing you've done, this is almost definitely a gcode issue. Most likely somewhere in there an M77 is being called, which stops the print timer. If you can't find this in there, feel free to upload your gcode and I can have a look.

Jyers, 2021-06-02:
Closing due to inactivity, it you have any updates on this feel free to let me know and I can reopen it.

closed 2021-06-02 by Jyers