Jyers/Marlin #934: Cannot start print from SD after preheat.

xantrk, 2021-05-30:
### Description

I usually, preheat, change filament, and go print menu from SD card. Selecting the file does not start the print. After a power cycle it works.
### Steps to Reproduce

1. Preheat
2. Change filament
3. (Optional) Cooldown, then preheat bed only (E heater sometimes affect probing)
4. Go to print from SD menu
5. Select a file, nothing happens
6. Power cycle, it works.

Cal-LF, 2021-05-30:
I have no issue with preheat then going to print the object. Just tried it. There is a delay of about 30 seconds as the temps align to the slicer settings. Try waiting a little longer for print to start.

xantrk, 2021-05-30:
> There is a delay of about 30 seconds as the temps align to the slicer settings. Try waiting a little longer for print to start.

I am stuck in list view, not printing progress screen and temps are not set. But never waited 30 seconds. Do you have the same?

Cal-LF, 2021-05-30:
The SD card may have failed.

xantrk, 2021-05-30:
> The SD card may have failed.

Unless I preheat and cooldown everything is fine even with back to back prints.

gothcha, 2021-05-31:
@xantrk I also checked this and have no issue, what firmware file are you using using? Also when you say step 2 change filament is this manually or using the filament change from menu?

xantrk, 2021-05-31:
> @xantrk I also checked this and have no issue, what firmware file are you using using? Also when you say step 2 change filament is this manually or using the filament change from menu?

Thanks for checking,  I am using filament change menu

Cal-LF, 2021-05-31:
This issue should be fixed the the latest commit. Heater timeout on Pause.

Jyers, 2021-05-31:
@Cal-LF I don't actually think these two are related as they aren't changing filament during the print, they are doing it before.

xantrk, 2021-05-31:
> @Cal-LF I don't actually think these two are related as they aren't changing filament during the print, they are doing it before.

Thanks @Jyers.
Might be related, my preheat is higher than initial SD print temperature. But I don't think it reads any gcode at all. 

Another thing might be related, I just re-enabled "SDCARD_SORT_ALPHA", and realized print list is much more responsive. It drops a lot of frames and almost hard to scroll with this flag disabled. Might also be a cause, that I was trying to select an item while it was not responsive.

Jyers, 2021-05-31:
Interesting, tia possible getting chronological info about the file day takes some extra processing. So is it printing like it should now?

xantrk, 2021-05-31:
I am unable to replicate the issue with the alphabetical sort. Thanks for the attention, I will update if it comes back.

closed 2021-05-31 by xantrk

gothcha, 2021-05-31:
Well this was an exciting evening, two issues sorted for the price of one!! :)